nightscout/AndroidAPS #3467: Equil pump driver

karzyg555, 2024-10-14:
I'm testing dev version and Equil pump integration. I've noticed 2 potential issues:
- When i try to inject single insulin dose equal or above 10 units i got communication problem between AndroidAPS and pump
- When insulin reservoirs will become empty and piston is going back, AndroidAPS app still shows progress of injection of insulin

MilosKozak, 2025-01-02:
Issue confirmed

MilosKozak, 2026-10-11:
Update, tested on a real Equil (test pump) with AAPS 4.0.0-dev (b5eb1bf8f1):

1. **Bolus of 10 U or more: fixed.** 10 U and 15 U (= max bolus) were accepted and delivered; the reservoir dropped by 10 U and 15 U. Details in #4137.
2. **Progress continues when the reservoir runs empty: still present.** `EquilManager.bolus()` shows progress from elapsed time only, and the delivered amount comes from that, not from the pump. A mid-bolus alarm cannot stop it, because the scan for pump adverts stops once connected. Keeping this issue open for that part.

MilosKozak, 2026-10-11:
The second part is fixed on dev in a9d4f76428, tested on a real Equil (test pump) by running a 12 U and a 4 U bolus into an empty reservoir.

What the pump does: it logs "Reservoir empty", stops all delivery (basal too) and rewinds the piston. AAPS showed the progress from a timer, recorded the full amount, and learned about it only at the next loop run, or even only after the refill.

Now:
- Right after a bolus AAPS connects to the pump again and reads its history (the pump closes the link about 10 s after the bolus command, so before nothing was read).
- "Reservoir empty" raises a pump alarm.
- If the reservoir ran empty during the bolus, only what was given until then is recorded (bolus speed x time, rounded up to 0.05 U). The progress dialog and an alarm show both amounts, for example "The pump stopped during the bolus. Given: 3.90 U of 4.00 U".
- From the empty moment until the refill AAPS records 0 U/h, so IOB does not count basal that was never given.
- The Equil screen has a Refresh button, which reads the pump status and history at once.

The first part (bolus of 10 U or more) was fixed before, see the comment above. Closing.

closed 2026-10-11 by MilosKozak