nightscout/AndroidAPS #3470: AAPS After database reset: loop process blocks, app crashes on manual calculation (tab SMB/AUTOISF - swipe)

vanelsberg, 2024-10-15:
**AndroidAPS 3.3.0-dev-c**
Flavor: fullDebug
Build commit# ba1c475fcce4a7275e21d5be3799aefb5771e0b6

Problem is when config is set to OpenAPS SMB or AutoISF
When selecting OpenAPS AMA all seems well

*After database reset:*
- Loop stops, no APS/algo calculations on SBM/AutoISF tab
- App crashes when APS calculation is forced by manual swipe on this tab

**Additional info:**
Just before crashing AAPS briefly show the "Fallback to SMB. Not enough TDD data"
This is even after 6++ hours...

*Logfile:*
[AndroidAPS.log](https://github.com/user-attachments/files/17384814/AndroidAPS.log)

*Logcat of crash after forcing loop calc:*
[image]

ga-zelle, 2024-10-15:
When testing new averageISs for AutoISF with dev-b I saw similar behaviour:
1. cleared database
2. cleared results to start with clean averages
3.  With 1 minute random BG the loop did not trigger on its own.
Every few minutes I pressed "run now" and after about 20-30 minutes it would run on its own. 

I did not investigate further because I had a different focus.

Philoul, 2024-10-16:
@ga-zelle Hum for me it's not the same behaviour, it looks like it take several minutes to have loop working in your side, but AAPS don't crash

@vanelsberg  crash immediatly so it can be difficult to recover a working situation (probably he will have to switch back to an old working branch, let it work for a few minutes to feed again database with records and then switch again to latest dev... (or something like that 🤔)

ga-zelle, 2024-10-16:
After updating to ladybug and merging all latest dev commits still working. Cleared database and results - still working after a pause of 1-2 minutes. This is full release in AS emulation.

vanelsberg, 2024-10-16:
> ... crash immediately so it can be difficult to recover ...

It crashes as soon as calculations are triggered (new BG or manual swipe). So I can switch to AMA to stop the crashing and starting loop. Aven after 24 hours of TDD, still same behavior.
Thinking I might have tinkered a bit to much with the database corrupting it, so tried uninstalling, restart phone, install+import settings. That did not solve the problem.

Currently on a branch so not latest. Will test tomorrow to see if latest dev changes solve the problem... 🤔

MilosKozak, 2024-11-13:
i tested it on SMB and got "No profile" result which is expected

MilosKozak, 2024-11-30:
still reproducible?

vanelsberg, 2024-11-30:
> still reproducible?

Will retry ..

vanelsberg, 2024-12-01:
@MilosKozak Retested with todays dev-e build.
Can not repro, feel free to close this issue.

closed 2024-12-01 by MilosKozak