nightscout/AndroidAPS #3471: Can't connect to Omnipod DASH pods after Android 15 upgrade

n0rbertg, 2024-10-16:
Phone: Pixel 9 Pro
Android version: Android 15 (stable)
Version: 3.2.0.4

After upgrading to Android 15, the app can't communicate with Dash. After rebooting the phone, a single handshake works but nothing after. Similarly, I was able to pair a new pod under Android 15, but any connection after times out. 

```03:54:15.543 [RxCachedThreadScheduler-63] D/AUTOMATION: [AutomationPlugin$onStart$11.accept():180]: Grabbed new BT event: EventBTChange[deviceAddress=AC:3B:96:37:EA:BB,deviceName=<null>,state=CONNECT]
03:54:15.548 [RxCachedThreadScheduler-63] D/AUTOMATION: [AutomationPlugin.processActions$automation_fullRelease():255]: processActions
03:54:15.552 [RxCachedThreadScheduler-63] D/AUTOMATION: [TriggerBg.shouldRun():62]: NOT ready for execution: Glucose is lesser than 3.8 MMOL
03:54:15.583 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
03:54:15.628 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 8
03:54:15.730 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 8
03:54:15.832 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 8
03:54:15.934 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 8
03:54:16.037 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 8
03:54:16.087 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
03:54:16.139 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 8
03:54:16.241 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 8
03:54:16.343 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:16.446 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:16.549 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:16.592 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
03:54:16.651 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:16.755 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:16.856 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:16.959 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:17.061 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:17.098 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
03:54:17.163 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:17.265 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 9
03:54:17.367 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:17.469 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:17.571 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:17.606 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
03:54:17.674 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:17.776 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:17.878 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:17.980 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:18.083 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:18.112 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
03:54:18.185 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:18.288 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 10
03:54:18.390 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:18.493 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:18.594 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:18.616 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
03:54:18.697 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:18.771 [binder:4453_F] D/PUMPBTCOMM: [BleCommCallbacks.onConnectionStateChange():38]: OnConnectionStateChange with status/state: 19/0
03:54:18.772 [binder:4453_F] I/PUMPBTCOMM: [Connection.onConnectionLost():209]: Lost connection with status: 19
03:54:18.799 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:18.902 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:19.007 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:19.110 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():102]: handshaking 11
03:54:19.119 [RxCachedThreadScheduler-67] D/PUMPBTCOMM: [Connection.disconnect():129]: Disconnecting closeGatt=false
03:54:19.124 [binder:4453_F] D/PUMPBTCOMM: [Connection.disconnect():129]: Disconnecting closeGatt=true
03:54:19.124 [RxCachedThreadScheduler-67] W/PUMPBTCOMM: [OmnipodDashManagerImpl$observeConnectToPodWithStop$1$1.accept():122]: observeConnectToPodWithStop error=info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: stopConnecting called
03:54:19.125 [RxCachedThreadScheduler-67] D/PUMP: [OmnipodDashManagerImpl$ErrorInterceptor.accept():780]: Intercepted error in OmnipodDashManagerImpl: info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: stopConnecting called
03:54:19.126 [binder:4453_F] D/PUMPBTCOMM: [BleCommCallbacks.resetConnection():217]: Reset connection
03:54:19.126 [ConnectionThread] I/PUMPCOMM: [OmnipodDashPumpPlugin$connect$2.invoke():276]: connect error=java.lang.RuntimeException: info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: stopConnecting called
03:54:19.213 [Thread-1378] D/PUMPQUEUE: [QueueThread.run():114]: connect
03:54:19.214 [Thread-1378] I/PUMP: [OmnipodDashPumpPlugin.connect():255]: connect reason=Connection needed
03:54:19.219 [RxCachedThreadScheduler-70] D/PUMP: [OmnipodDashManagerImpl$PodEventInterceptor.accept():704]: Intercepted PodEvent in OmnipodDashManagerImpl: BluetoothConnecting
03:54:19.223 [RxCachedThreadScheduler-70] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 0
03:54:19.224 [RxCachedThreadScheduler-70] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@af28aee[Count = 1])
03:54:19.232 [RxCachedThreadScheduler-70] D/PUMPBTCOMM: [Connection.waitForConnection():145]: waitForConnection connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@af28aee[Count = 1])
03:54:19.287 [RxCachedThreadScheduler-63] D/AUTOMATION: [AutomationPlugin$onStart$11.accept():180]: Grabbed new BT event: EventBTChange[deviceAddress=AC:3B:96:37:EA:BB,deviceName=<null>,state=DISCONNECT]
03:54:19.290 [RxCachedThreadScheduler-63] D/AUTOMATION: [AutomationPlugin.processActions$automation_fullRelease():255]: processActions
[...]
07:09:13.225 [CommandQueueImplementationHandler] D/PUMPQUEUE: [CommandQueueImplementation.notifyAboutNewCommand$lambda$6():220]: Starting new thread
07:09:13.227 [RxCachedThreadScheduler-66] D/CORE: [UserEntryLoggerImpl$log$1.accept():60]: USER ENTRY: 10/16/24 07:09:13 TEMP_BASAL Loop  [UnitPerHour(value=4.7), Minute(value=30)]
07:09:13.230 [Thread-2559] D/PUMPQUEUE: [QueueThread.run():114]: connect
07:09:13.230 [Thread-2559] I/PUMP: [OmnipodDashPumpPlugin.connect():255]: connect reason=Connection needed
07:09:13.231 [RxCachedThreadScheduler-77] D/PUMP: [OmnipodDashManagerImpl$PodEventInterceptor.accept():704]: Intercepted PodEvent in OmnipodDashManagerImpl: BluetoothConnecting
07:09:13.233 [RxCachedThreadScheduler-77] D/PUMPBTCOMM: [Connection.connectionState():171]: GATT connection state: 2
07:09:13.233 [RxCachedThreadScheduler-77] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@413994e[Count = 1])
07:09:13.235 [RxCachedThreadScheduler-77] D/PUMPBTCOMM: [Connection.waitForConnection():145]: waitForConnection connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@413994e[Count = 1])
07:09:13.236 [binder:4453_2] D/PUMPBTCOMM: [BleCommCallbacks.onConnectionStateChange():38]: OnConnectionStateChange with status/state: 0/2
[... cut ...]
07:09:13.753 [RxCachedThreadScheduler-77] W/PUMPBTCOMM: [OmnipodDashManagerImpl$observeConnectToPodWithStop$1$1.accept():122]: observeConnectToPodWithStop error=info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: stopConnecting called
07:09:13.753 [RxCachedThreadScheduler-77] D/PUMP: [OmnipodDashManagerImpl$ErrorInterceptor.accept():780]: Intercepted error in OmnipodDashManagerImpl: info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: stopConnecting called```

2flea, 2024-10-16:
Same issue on Pixel 8 with Android 15. After reboot the first connection works but then no handshake possible. Switched to another phone using < A15 for now. 
@MilosKozak please forward to Omnipod specialists. 
Cheers, 
Andreas

mariogarofalo, 2024-10-17:
Even I same bug.
Pixel 8 with Android 15 and Omnipod Dash.
However, it seems that, despite the connection error, the basal rate adapts based on the glucose level (increases or decreases). SMBs are also delivered, except when I try to record my carbohydrate intake following a meal, the bolus is not delivered.
I hope for a bugfix for the app as soon as possible, because it's a really big problem.

Larsboyscout, 2024-10-17:
I had it connect once dash g6 and it delivered insulin

woernsn, 2024-10-17:
Same issue here on Pixel 7 after upgrade to Android 15.
I switched to an older phone for AAPS in the meantime.

jtmarcot, 2024-10-18:
I'm holding off on Android 15 until this is confirmed to be sorted. Thanks for opening the issue, and thanks to anyone working on it!

3.2.0.4 on Pixel 7

2flea, 2024-10-19:
Some logs from 3 days ago. 

08:30:47.923 [RxCachedThreadScheduler-34] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@232d152[Count = 1])
08:30:49.664 [DefaultDispatcher-worker-2] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.receivers.KeepAliveWorker
08:30:54.220 [RxCachedThreadScheduler-36] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@471996[Count = 1])
08:31:19.141 [RxCachedThreadScheduler-48] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@8d7c097[Count = 1])
08:31:24.407 [RxCachedThreadScheduler-33] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@bec7c86[Count = 1])
08:31:38.189 [main] D/WIDGET: [Widget.onReceive():87]: onReceive ScheduleEveryMin
08:31:38.554 [CompanionHandler] D/CORE: [Widget.updateTemporaryTarget():223]: Adjusted target. Profile: 100.0 APS: 93.0
08:31:46.488 [RxCachedThreadScheduler-34] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=null, stopConnection=java.util.concurrent.CountDownLatch@c13e9d2[Count = 1])
08:31:52.623 [RxCachedThreadScheduler-36] D/CORE: [Connection.connect():69]: Connecting connectionWaitCond=ConnectionWaitCondition(timeoutMs=30000, stopConnection=null)
08:32:22.660 [RxCachedThreadScheduler-36] E/PUMP: [OmnipodDashPumpPlugin$executeProgrammingCommand$3.accept():1364]: Error executing command
info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: Service not found: 1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.ServiceDiscoverer.discoverServices(ServiceDiscoverer.kt:54)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.session.Connection.connect(Connection.kt:101)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl.connect$lambda$1(OmnipodDashBleManagerImpl.kt:154)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl.$r8$lambda$c2IZX_V0B5Ile126GNiSOANZRgY(Unknown Source:0)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl$$ExternalSyntheticLambda0.subscribe(Unknown Source:4)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:41)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDefer.subscribeActual(ObservableDefer.java:40)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.drain(ObservableConcatMap.java:469)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.onSubscribe(ObservableConcatMap.java:330)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableFromArray.subscribeActual(ObservableFromArray.java:32)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap.subscribeActual(ObservableConcatMap.java:56)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
	at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
	at java.lang.Thread.run(Thread.java:1012)

What exactly is Service: 1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f?

Does anyone know?

mariogarofalo, 2024-10-20:
Do you have information on the work status of a possible bug fix for this problem? 
My glucose levels have been through the roof for almost a week now

tyrex1975, 2024-10-20:
> Do you have information on the work status of a possible bug fix for this problem? My glucose levels have been through the roof for almost a week now

This is a volunteer project and nobody is required to offer solutions or timeframes. I'm sure the developers are now aware of the problem. In the meantime I suggest you revert to other management practices if AAPS is not working for you, we are all responsible for our own care.

n0rbertg, 2024-10-20:
> Some logs from 3 days ago.

I also have similar in the logs

```
07:01:43.780 [RxCachedThreadScheduler-74] W/PUMPBTCOMM: [OmnipodDashManagerImpl$observeConnectToPod$1$1.accept():129]: observeConnectToPod error=info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: Service not found: 1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f
07:01:43.781 [RxCachedThreadScheduler-74] D/PUMP: [OmnipodDashManagerImpl$ErrorInterceptor.accept():780]: Intercepted error in OmnipodDashManagerImpl: info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: Service not found: 1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f
07:01:43.782 [RxCachedThreadScheduler-74] E/PUMP: [OmnipodDashPumpPlugin$executeProgrammingCommand$3.accept():1364]: Error executing command
info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.exceptions.ConnectException: Service not found: 1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.ServiceDiscoverer.discoverServices(ServiceDiscoverer.kt:54)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.session.Connection.connect(Connection.kt:101)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl.connect$lambda$1(OmnipodDashBleManagerImpl.kt:154)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl.$r8$lambda$c2IZX_V0B5Ile126GNiSOANZRgY(Unknown Source:0)
	at info.nightscout.androidaps.plugins.pump.omnipod.dash.driver.comm.OmnipodDashBleManagerImpl$$ExternalSyntheticLambda0.subscribe(Unknown Source:4)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableCreate.subscribeActual(ObservableCreate.java:41)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDefer.subscribeActual(ObservableDefer.java:40)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.drain(ObservableConcatMap.java:469)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap$ConcatMapDelayErrorObserver.onSubscribe(ObservableConcatMap.java:330)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableFromArray.subscribeActual(ObservableFromArray.java:32)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableConcatMap.subscribeActual(ObservableConcatMap.java:56)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableDoOnEach.subscribeActual(ObservableDoOnEach.java:42)
	at io.reactivex.rxjava3.core.Observable.subscribe(Observable.java:13262)
	at io.reactivex.rxjava3.internal.operators.observable.ObservableSubscribeOn$SubscribeTask.run(ObservableSubscribeOn.java:96)
	at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:644)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
	at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
	at java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:307)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1145)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:644)
```

mariogarofalo, 2024-10-21:
Here, mine

[AndroidAPS.log](https://github.com/user-attachments/files/17456680/AndroidAPS.log)

2flea, 2024-10-21:
@nickb24 I exported some logs with logcat. Let me know if you need more.

[Google-Pixel-8-Android-15_2024-10-21_100641.logcat.zip](https://github.com/user-attachments/files/17457689/Google-Pixel-8-Android-15_2024-10-21_100641.logcat.zip)

t1dude, 2024-10-21:
FYI it looks like Dexcom is having Bluetooth issues with Android 15 as well: https://www.dexcom.com/en-AU/faqs/will-android-15-affect-my-dexcom-g7-cgm-system

osodebailar, 2024-10-21:
> FYI it looks like Dexcom is having Bluetooth issues with Android 15 as well: https://www.dexcom.com/en-AU/faqs/will-android-15-affect-my-dexcom-g7-cgm-system

Yes they have , because ITS a Google issue, because of changing the BT stack in Android 15

vanelsberg, 2024-10-21:
> What exactly is Service: 1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f?
> Does anyone know?

The error occurs in the code where the DASH driver is scanning for a BT device with a specific BT service ID _1a7e-4024-e3ed-4464-8b7e-751e03d0dc5f_ to connect to.

And yes, this ID is fixed in the code because it is unique for DASH BT devices.

Btw: We have seen this problem in the past during DASH driver development with some phone/brands where the DASH BT Firmware was somehow incompatible with the Phone's BT implementation. Other reason could be the OS blocking BT access due to security settings or dependencies?

Not sure if this can be solved though code...

Suggestion: Try the nRF Connect tool app (Playstore): See if does find a DASH device with this service ID?
- If it does not: go find some setting in the Phone OS?
- If it does there is at least there is a chance AAPS should be able to connect in same way...

2flea, 2024-10-22:
nRF connect tool finds the device with ever scan.

[image]
[image]
[image]
[image]

vanelsberg, 2024-10-22:
> I couldn't find anything in the logs. This looks like an Android OS issue.
> Based on this post that was closed today: https://issuetracker.google.com/issues/362191722#comment7

Quoting this issue:
_We have an intermittent but easily reproducible issue that occurs with only one of our BLE device series. Connecting to the device becomes impossible because the connection gets stuck during service discovery.  Once the issue occurs, all next connection attempts fail for the same reason. Power cycling the device or restarting the app doesn't help. The issue disappears after toggling Bluetooth OFF and ON on the phone. The problem is likely related to BLE chipset or firmware and Android compatibility, as only one particular device series is affected._

Based on what you all have comment here, it all looks like this issue.
Unless there is some work around in the AAPS driver code, I'd say this is a compatibility issue to be fixed in the OS 🤔

vanelsberg, 2024-10-22:
> Seems to be the same error reported in this bug that was closed due to inactivity: #935
Old & I don't think so. Other then: yes: some phones brands/types just are not compatible with DASH. This still goes.
But in this case it looks more like this is related to Android 15, not a specific phone brand/type?

woernsn, 2024-10-22:
> Based on what you all have comment here, it all looks like this issue. Unless there is some work around in the AAPS driver code, I'd say this is a compatibility issue to be fixed in the OS 🤔

I would go so far to speculate that also the Dexcom issue is the same.
Dexcom stated that they are in contact with Google directly. Therefore I guess, an Android update will need to happen soon .

2flea, 2024-10-22:
It looks like that QPR1 Beta 2 fixed this behaviour. Mario tested it yesterday evening and during the night. I wait for Beta 3 OTA as I do not wish to wipe my data atm. Let's hope that Google's fix works

n0rbertg, 2024-10-22:
> Can someone with the issue try installing Android 15 QPR1 Beta 2?

I installed last night and during the night it was stable, without signal losses.  I am able to connect to the pod now. 

So it is an Android issue as it seems.

2flea, 2024-10-22:
> I suggest an admin for the Facebook group post an announcement warning Pixel users not to upgrade to Android 15. I have a Pixel 7 pro and I'm holding off on it (although I didn't get the notification on my phone yet). I can't imagine updating one day and all of a sudden my loop system is broken. In my case, I don't even have my Omnipod PDM setup so I would be in big trouble if I need to scramble to get another phone setup just so I can continue to run my pump.

I asked Tim Street on FB

2flea, 2024-10-22:
> To debug further though, I would need a volunteer with the issue to do bluetooth logging as described by Google here: https://issuetracker.google.com/issues/362191722#comment7
> 
> ```
> 1. Go to Developer Settings ->  and set Enable Bluetooth HCI snoop log to Enabled.
> 
> 2. Also increase the log buffer size by following these steps : System settings-> Developer options-> Logger buffer sizes-> 8M
> 
> Then capture and share a new bug report.
> ```

Where will I find the log file on my Pixel? 
If you want I can try to get some HCI logs.

Btw, the app nRF Connect also does not discover any services of the Dash. It has the same behaviour than AAPS. It connects but once it calls discoverServices there is none found but it looks like a timeout.

2flea, 2024-10-23:
> > > To debug further though, I would need a volunteer with the issue to do bluetooth logging as described by Google here: https://issuetracker.google.com/issues/362191722#comment7
> > > ```
> > > 1. Go to Developer Settings ->  and set Enable Bluetooth HCI snoop log to Enabled.
> > > 
> > > 2. Also increase the log buffer size by following these steps : System settings-> Developer options-> Logger buffer sizes-> 8M
> > > 
> > > Then capture and share a new bug report.
> > > ```
> > 
> > 
> > Where will I find the log file on my Pixel? If you want I can try to get some HCI logs.
> > Btw, the app nRF Connect also does not discover any services of the Dash. It has the same behaviour than AAPS. It connects but once it calls discoverServices there is none found but it looks like a timeout.
> 
> You need to be careful sharing these logs. It will contain sensitive information. More information: https://developer.android.com/studio/debug/bug-report https://source.android.com/docs/core/connect/bluetooth/verifying_debugging#debugging-with-logs
> 
> To remove sensitive information you would need to parse the bug report and only share the bluetooth log.

Not sure if this helps and if the Dash request is in it at all - I only switched AAPS back on for 120sec till timeout and enabled HCI just before. Let me know if you find anything except my Garmin watch ;-) (I think there are no sensitive data inside, please let me know if there are any).

[btsnooz_hci.log](https://github.com/user-attachments/files/17486927/btsnooz_hci.log)

2flea, 2024-10-23:
Log Buffer size is/was 1M as suggested by Google.

2flea, 2024-10-23:
> > > > > To debug further though, I would need a volunteer with the issue to do bluetooth logging as described by Google here: https://issuetracker.google.com/issues/362191722#comment7
> > > > > ```
> > > > > 1. Go to Developer Settings ->  and set Enable Bluetooth HCI snoop log to Enabled.
> > > > > 
> > > > > 2. Also increase the log buffer size by following these steps : System settings-> Developer options-> Logger buffer sizes-> 8M
> > > > > 
> > > > > Then capture and share a new bug report.
> > > > > ```
> > > > 
> > > > 
> > > > Where will I find the log file on my Pixel? If you want I can try to get some HCI logs.
> > > > Btw, the app nRF Connect also does not discover any services of the Dash. It has the same behaviour than AAPS. It connects but once it calls discoverServices there is none found but it looks like a timeout.
> > > 
> > > 
> > > You need to be careful sharing these logs. It will contain sensitive information. More information: https://developer.android.com/studio/debug/bug-report https://source.android.com/docs/core/connect/bluetooth/verifying_debugging#debugging-with-logs
> > > To remove sensitive information you would need to parse the bug report and only share the bluetooth log.
> > 
> > 
> > Not sure if this helps and if the Dash request is in it at all - I only switched AAPS back on for 120sec till timeout and enabled HCI just before (incl. BT off/on). Let me know if you find anything except my Garmin watch ;-) (I think there are no sensitive data inside, please let me know if there are any).
> > [btsnooz_hci.log](https://github.com/user-attachments/files/17486927/btsnooz_hci.log)
> 
> I'm taking a look at the log now. There is no personal information in it. However all the Garmin packets are cluttering the logs so it will take me some time.

Wait ... I will switch off my watch and redo the test for 5min or so...

2flea, 2024-10-23:
You're right, I read 1M elsewhere for debug setting, not verbose.
New data with log buffer 8M and Garmin switched off
[btsnooz_hci.log](https://github.com/user-attachments/files/17493068/btsnooz_hci.log)

ahantschel, 2024-10-23:
As i mentioned above in Post #2 i had the same issues on the Android 15 Beta.

Now i wanted to test it in the Final and after that Install QPR Beta 2.
So i did the OTA Update 14 --> 15 and it now works without any problems for me. (On 15 final not QPR Beta)

I had a Dash Pod active yesterday during the Update and also started a new one later the evening.

Now i am running AAPS for over 24h under Android 15 without any connection issues.

Very strange....

2flea, 2024-10-23:
> As i mentioned above in Post #2 i had the same issues on the Android 15 Beta.
>
> Now i wanted to test it in the Final and after that Install QPR Beta 2. So i did the OTA Update 14 --> 15 and it now [...]

Which phone and which Build number do you have?

ahantschel, 2024-10-23:
Pixel 6a
Android Build AP3A.241005.015
AndroidAPS 3.2.0.4

[image]

[image]

2flea, 2024-10-23:
> > Can someone with the issue try installing Android 15 QPR1 Beta 2?
> 
> I installed last night and during the night it was stable, without signal losses. I am able to connect to the pod now.
> 
> So it is an Android issue as it seems.

I installed QPR1 Beta 3 OTA tonight and just tested with nRF. It can connect to Dash and discover services again. 

[image]

2flea, 2024-10-23:
AAPS is also back to normal with QPR1 Beta 3.

Larsboyscout, 2024-10-24:
> I got my hands on a Pixel 6 to test. I flashed it to 15.0.0 (AP3A.241005.015, Oct 2024). I can't reproduce the issue reported here. Perhaps this only effects Pixel 7 and above?
https://github.com/nightscout/AndroidAPS/issues/3471#issuecomment-2420216457 this comment above is using a 6A same gen as the 7 i assume so yes it looks like that gen and above

CJiwan, 2024-10-28:
I have installe the QPR1 Beta 3 (Build : AP41.240925.009) on my Pixel 7a.
Unfortunately I still have the same problem. I tried restarting the phone but it doesn't change anything
For info I use the Dexcom G6 with Byoda and omnipod dash.

2flea, 2024-10-28:
I did the OTA to QPR1 Beta 3 and also still had issues. Then I did a flash (using Android flash tool) with QPR1 Beta 3 and all seemed ok. But then out of nowhere I had issues again that AAPS nor nRF could discover services. Even restart of the phone did not work. This happened several times with no obvious correlation whatsoever. Handshaking when working took usually 1-3sec, with the "situation" it ended up in timeout. I even could not give a manual bolus for my cake ;-)

I ended up flashing back to stable Android 14 and will stay there. 

Here my AAPS log from the time it happened yesterday afternoon 15:36 - 15:54. I only enabled ALL pump related logs, hope it helps... as I said I am back to 14 as this state of 15 is unreliable and unstable.
[AndroidAPS.log](https://github.com/user-attachments/files/17538079/AndroidAPS.log)

2flea, 2024-10-28:
> @2flea Can you try again with nRF connect? Can you put AAPS in virtual pump mode, reboot phone, force stop AAPS, then try with nRF connect?

At that particular time I checked also with nRF and it could also not discover the service.
The discover suddenly worked again for nRF and AAPS. 

That is the reason I am now back to A14 and can no longer reproduce anything, sorry.

fawkes15, 2024-10-29:
Hi all,
I'm on Pixel8 A15 for two weeks and I never had any issues with DASH and BT. Let me know how can I help you...

p.s. I've had issue with Dexcom G7 during warm up pairing (2 sensors fail)

alarictabaries, 2024-10-30:
Hi everyone,

I mistakenly updated my Pixel 9 to Android 15 a bit too early.

I believe I read somewhere that the beta updates might help fix the issue – any thoughts on that?

Thanks for the incredible work on this tool!

n0rbertg, 2024-10-30:
> I mistakenly updated my Pixel 9 to Android 15 a bit too early.
> 
> I believe I read somewhere that the beta updates might help fix the issue – any thoughts on that?

Hi,

I have a Pixel 9 Pro and upgrading to QPR1 beta2 and beta3 solved the issue for me. It's stable now both with Dash and G7 (via Diakem). 

Other people still have issues here after upgrading to QPR1 beta, and based on the limited amount of data it's not confirmed if it's model specific or caused by something else. In my case, upgrading to beta worked.

alarictabaries, 2024-10-30:
> @alarictabaries Do you have an issue on your Pixel 9 with Android 15? If yes, you can try signing up for the Android Beta and see if it resolves your problem. Please report back if you do: https://www.google.com/android/beta#devices

> Other people still have issues here after upgrading to QPR1 beta, and based on the limited amount of data it's not confirmed if it's model specific or caused by something else. In my case, upgrading to beta worke

Thank you for your answers! I’ll test it with my new Dash on Friday night and will keep you updated.

alarictabaries, 2024-11-01:
I’ve updated my Pixel 9 to the latest beta version, but unfortunately, the issue still persists.

alarictabaries, 2024-11-02:
> Please report back. If indeed the latest beta doesn't resolve your issue, then it means that Android 15 is not going to work with Dash pods for awhile.

I confirm the issue is still here for me. I tested a pod at 22:49, and I’m including the logs here in case it helps with troubleshooting.

- [AndroidAPS._2024-11-02_00-25-16_.1.txt](https://github.com/user-attachments/files/17605089/AndroidAPS._2024-11-02_00-25-16_.1.txt)

- [AndroidAPS._2024-11-02_00-25-16_.0.txt](https://github.com/user-attachments/files/17605088/AndroidAPS._2024-11-02_00-25-16_.0.txt)

alarictabaries, 2024-11-03:
> 1. Did you have good connection all the time when the phone was plugged in? Looking to see if some battery saving if causing issues.

This could indeed be a cause ! I hadn't paid attention to it.

> 2\. One log looks completely good. The other has a connection issue that looks to be resolved immediately after you restarted AAPS right? If this is true then it suggests an AAPS code issue and not an OS bluetooth stack issue.

It was somewhat inconsistent, sometimes it worked after restarting the phone, sometimes after restarting the app, and sometimes it didn’t work at all.

I downgraded to Android 14, I can't help you further with this issue.

Larsboyscout, 2024-11-04:
I can test i am running 2 phones I've pixel 7 with Android 15 and a pixel 2
with advertised 14

2flea, 2024-11-04:
I can't no longer help with Android 15 but if it helps or as a control I could do that with Android 14.

tyrex1975, 2024-11-04:
I have a Pixel 6 which is my daily driver for AAPS so I plan on staying on Android 14 for the time being. However, my wife has a Google Pixel 7a which I can get upgraded to Android 15 and run a patched AAPS version. I have spare Dash pods which could be used separately without messing up my main setup. Will need some help in Android Studio to build the patched version and then returning to the main branch for my daily driver AAPS afterwards though....

mariogarofalo, 2024-11-06:
Novembre changelog for Pixel phones:

The biggest updates are for the Pixel 9, 9 Pro, 9 Pro XL, and 9 Pro Fold (AP3A.241105.008):

Security: Provides the most up to date Android security patches on your device.

Bluetooth: Addressed an issue with Bluetooth range under certain conditions.

Camera: Addressed an issue with camera tilt when zooming between cameras under certain conditions.

Sensors: Addressed an issue that occasionally prevented Adaptive brightness from activating in certain conditions.

Touch: Addressed an issue when pressing the keyboard dismiss button in certain conditions.

User interface: General improvements for performance and stability in certain UI transitions and animations.

Display & Graphics: Resolved an issue that caused white dots to flash under certain conditions.

Meanwhile, there are just two November 2024 update changelog entries for the Pixel 6, 7, and 8 series (AP3A.241105.007):

The most up to date Android security patches on your device.
General improvements for performance and stability in certain UI transitions and animations.

mhuang002, 2024-11-10:
On Android 15 beta, pixel 7. I was having a ton of connection issues with omnipod dash and AAPS. Would connect only intermittently and rarely.

I only could bolus after I fully rebooted my phone. Even then, the bolus would often stop after maybe 3 units. 

Then I turned off Play Protect in the play store, and my omnipod connected again. So far so good. 

Hope this helps someone else.

mariogarofalo, 2024-11-10:
I've followed Mike Huang suggestion.

I've filled a new pod with water and put on my left leg.
I've deactivate Play Protect from Play Store.
Pod started.
The bolus launched from Aaps from phone, was delivered.
The bolus launched from Aaps from Samsung Galaxy Watch 6, gone in timeout.
But this Is a greet result for me.
Tomorrow evening my actual pod, connected with old phone, expire. I return to use my main phone (Pixel 8 with Android 15) as ever.

fawkes15, 2024-11-10:
I'd like to report that after the November security patch (5 november), I no longer have pairing issues with the Dexcom G7 and Pixel Android 15. 
I've never had issues with the Omnipod Dash since upgrading to A15

parapenT1sta, 2024-11-10:
About Play Protect, if it is on Wiki is for a reason: https://androidaps.readthedocs.io/en/latest/UsefulLinks/FAQ.html#aaps-stops-everyday-around-the-same-time

jchester47, 2024-11-10:
Disabling play protect did not resolve the problem for me. So that doesn't seem to be a viable fix.

I can generally get the connection to restore and hold by force closings AAPS, clearing the app's cache, and turning off Bluetooth. Then I reopen AAPS, and allow the app to turn Bluetooth back on.

The connection restores and usually holds until the next phone reboot or new pod pairing, although the connection is occasionally lost during the session.

Larsboyscout, 2024-11-11:
For me a restart fixes the connection but then it loses connection after a
while and I have to stop the alarm and restart to do it again

Larsboyscout, 2024-11-11:
And not sure if it matters but when I have. Suggestion for a Badal
adjustment the icon changes as per below

t1dude, 2024-11-11:
I chose to update my Pixel 9 Pro from Android 14 to Android 15 with the November update. I've had no issues with Dexcom G7 or Omnipod Dash so far. The pod that was already connected kept working well. I did a pod change and it worked well also.

t1dude, 2024-11-12:
No connection issues during the night either.

CJiwan, 2024-11-13:
Hello,
Has anyone with the problem had the opportunity to test the QPR2 Beta 1 which is available?

Heike2109, 2024-11-15:
After installing Android 15 I have issues with Omnipod Dash. After reading your commets, I installed nRF connect tool. There I bonded the Omnipod Dash and choose "Use autoConnect". Works fine. After changing the pod, you have to bond the new Pod.

CJiwan, 2024-11-15:
> After installing Android 15 I have issues with Omnipod Dash. After reading your commets, I installed nRF connect tool. There I bonded the Omnipod Dash and choose "Use autoConnect". Works fine. After changing the pod, you have to bond the new Pod.

This is interesting, next Monday I will try your solution when changing my POD.
If there are several of us for whom this solves the problem, it could perhaps help to find a correction in AAPS.

tyrex1975, 2024-11-16:
https://support.google.com/pixelphone/thread/305744117/google-pixel-update-november-2024?hl=en

Mentions:

Bluetooth
Fix for issue with Bluetooth range under certain conditions*[3]

*[3] Pixel 9, Pixel 9 Pro, Pixel 9 Pro XL, Pixel 9 Pro Fold

jchester47, 2024-11-16:
> Small update for the group. I've got my hands on a Pixel 7 (non-pro) running Android 15 November update (AP3A.241105.007). I connected AAPS to a running Pod and everything is working. Monitored with Logcat and there is nothing unusual there either. I refreshed the Pod status for the last 15 minutes and it works every time. I also sent a bolus command and it worked. 
> 
> For those who have issues, are you not able to refresh the Pod status every time? Or do you only notice random connection losses after a certain amount of time (like overnight)? 
> 
> Working theories:
> 
> 1. The November update fixed the problem,
> 2. Pixel 7 series is unaffected by the issue, and it's Pixel 8/9 the problem,
> 3. I installed AAPS using ADB, bypassing any Play Protect interferences possibly.
> 4. Issue is random and only occurs after a certain amount of time
> 
> I have to return the phone back to the person I borrowed it from, so I can't continue extended tests. My personal phone is a Pixel 7 Pro, but I'm not comfortable yet to upgrade it to Android 15.  

I am running on Android 15 beta on a Pixel 8 Pro (the latest version plus the 11/5 security update). I opted into Android 15 when the stock 15 update completely broke my ability to communicate with pods. 

On the beta version, I can connect to pods, but it is spotty. It usually goes like this:

1. I can prime, pair and insert a new pod.
2. The initial pairing and command for boluses is successful.
3. After the initial connection, no further connection attempts work. 
4. I delete my AAPS logs and force close AAPS. I clear the app cache. 
5. I turn off Bluetooth and go into my apps and clear the cache and data on the "bluetooth legacy" app. 
6. Upon reopening APS, i allow it to re-enable BT. 
7. After this, the connection to the pod is usually successful and persists until the next phone reset, at which time the problem recurs.

This isn't always consistent though. Sometimes the connection is randomly lost in the middle of a session (especially annoying in the middle of the night). Sometimes I just can't reestablish the connection no matter what I do, in which case I just have to give up on the pod and start over.

[164 more replies]