nightscout/AndroidAPS #3483: Approving temp basal crashes the app

YishaiBerg, 2024-10-20:
Approving temp basal suggestions crashes the app. I have completed objective 3, but in hindsight, it was happening also when trying to approve the suggestions that appeared before completing objective 3. I started with a virtual pump, and at some point moved to Omnipod Dash. Setting temp basals manually does seem to work. 

Looks like it is related to #3083 . 
The following line from the logs seems to reflect (one of) the precision times of the bug. If there is reason to believe otherwise I can verify it again.
`04:35:39.922 [OverviewFragmentHandler] D/APS: [LoopPlugin.invoke():232]: invoke from Accept temp button`   

- Build: 3df83535f9-2024.09.24
- Flavor: fullrelease
- NS version: 15.0.2
- Using with Xdrip+ and Omnipod Dash
- Phone model: Galaxy A25 5G
- Android 14
- Log files are attached
[AndroidAPS._2024-10-20_03-11-07_.1[1].zip](https://github.com/user-attachments/files/17449553/AndroidAPS._2024-10-20_03-11-07_.1.1.zip)
[AndroidAPS._2024-10-20_04-45-05_.2[1].zip](https://github.com/user-attachments/files/17449554/AndroidAPS._2024-10-20_04-45-05_.2.1.zip)

Thanks to all the contributors for the amazing work on the app! Will very much appreciate any help with this issue.

YishaiBerg, 2024-10-22:
Update - for some strange reason this only happens when using a PIN or custom password for bolus protection. When using biometrics or no protection it works. 
Is anyone able to reproduce this?

MilosKozak, 2024-11-04:
fixed by https://github.com/nightscout/AndroidAPS/commit/dadd409f105844f76ee08dc0ed136b6341dfe792

closed 2024-11-04 by MilosKozak