nightscout/AndroidAPS #3484: Add BG accel to BG delta list (copy from AutoISF to main build)

MowgliChild, 2024-10-21:
Functional improvement request: Add BG acc to main build

BG accel is displayed in AutoISF on top of the 3 normal deltas and used for various functions within this dev branch.

This is an excellent tool for seeing what is happening in real time, the code already exists and the real estate is already there on the Home Screen

[image]

ga-zelle, 2025-01-19:
It is a very useful information and easy to add. However, I am afraid users not familiar with AutoISF would have trouble understanding what it means.

MowgliChild, 2025-01-19:
> It is a very useful information and easy to add. However, I am afraid users not familiar with AutoISF would have trouble understanding what it means.

@ga-zelle Thanks for your comment. 

Acceleration belongs to basic physics, not autoisf.
If one understands what delta (velocity) is, then one should also understand what acceleration is.

Yes I think a small explanation in RTD may be helpful for both. 

It's s low 'cost' improvement.

koelewij, 2025-01-29:
It's not bad to add it to the UI.
Then it is nice to see it at a certain moment, but besides that it doesn't help users a lot if the loop does nothing with the information. 
Looking at the graph I have an idea of the acceleration also. 

Nevertheless it doesn't hurt to add it.