nightscout/AndroidAPS #3490: Medtrum Nano / stop bolus gives error

robertrub, 2024-10-23:
Last dev. Systematically, if I stop an ongoing bolus, I get a pump unreachable alert.

Either stopping the bolus messes with the BT connection, or triggers a wrong alert message. In both cases, it is a bug.

Logs are below. Check around 18:00

[image]

jbr7rr, 2024-11-13:
Looking into this one, requested more info from Robby at discord

robertrub, 2024-11-13:
Ok, correction, it was a partial user error. The text says check amounts but the 3 buttons look as it was a pump unreachable alert...

[image]

jbr7rr, 2024-11-13:
Clarified the texts in:
https://github.com/nightscout/AndroidAPS/pull/3529

In addition we may want to have a separate error dialog for bolus with only an OK button @MilosKozak any thoughts on this?

MilosKozak, 2024-11-13:
what should be there?

robertrub, 2024-11-13:
@MilosKozak  Actually, the message says check IOB (and COB with the last commit) but shows the same 3 buttons as a connection problem. The first one is mute but there are no sounds to mute, the 2nd is wait 5 min...

Only an OK button is needed under the "check amount of IOB & COB delivered" message.

Having the 3 buttons (and a _failed_ text) of the connection problems is confusing.

[image]

Philoul, 2025-11-04:
Fixed with #4283 (if the bolus was canceled by the user, it should should return success, it's done that way within Insight Driver).

@jbr7rr, this is my first contribution to Medtrum (could you check my PR please ?).