nightscout/AndroidAPS #3492: Negative carbs ?

robertrub, 2024-10-24:
Yesterday afternoon, I set a negative carbs entry as the big carbs entered in the morning were still not finished around 14:00

[image]

After the negative carbs entered,  I could bolus normally.

This morning, I see negative carbs on the main screen.... 

[image]

The negative carbs entered should "erase" carbs of COB but should not generate negative carbs.

MilosKozak, 2024-11-01:
I cannot reproduce this issue

robertrub, 2024-11-01:
I don't use neg carbs so can't tell if it happens regularly.

Will try to trigger it again.

Ii has set neg IOB to clear the 15gr to 0 then add the meal carbs and bolus. Some hours later, I saw the neg carbs.

vanelsberg, 2024-11-02:
_Confirming_: notices the same. No details on repro.
Will try to trigger again..

robertrub, 2024-11-02:
@MilosKozak It happened again. I have the screenshot the treatments tab (if it can help reproduce).

I saw the negative COB at 19h10 but it probably appeared earlier.

[image]

[image]

[image]

olorinmaia, 2024-11-03:
It would make more sense to "edit" a carb entry that is overestimated than actually adding a negative carb entry at a given time. It needs to be supported in NS v1/v3 api to sync correctly.

It is possible to edit directly in NS > Reports > Treatments, but it doesn't sync back to AAPS/AAPSClient. What I do today is to delete the carb entry that is wrong, and add a new entry with about same timestamp with correct number of carbs.

robertrub, 2024-11-04:
Normally I don't use neg carbs. I'm just testing the function. It seems that some people had a need for neg carbs though. It is faster to enter neg carbs than delete the carbs, enter new carbs with the old time stamp.

olorinmaia, 2024-11-04:
Yes, I am testing it as well. But I wonder if it is possible to add a function to edit instead within AAPS / AAPSClient.

vanelsberg, 2024-11-04:
> It would make more sense to "edit" a carb entry that is overestimated.

yes & no?
When I overestimated carbs I remove and then add the actual amount of carbs. So similar to "editing" as you propose.

But I think the original intention of negative carbs was not so much "editing" for "over/underestimated" carbs, but a way to handle situations where _carbs absorption is to slow_, without changing original amount of carbs registered.
That is a feature that I find extremely useful in certain situations.

robertrub, 2024-11-04:
Yes, "eliminating" slow COB reduction is where I use/test the option too.

MilosKozak, 2024-11-13:
try with https://github.com/nightscout/AndroidAPS/commit/9f3cbe21505da570054f06f86dc8f2b0c6841d9a

robertrub, 2024-11-13:
Will build this afternoon and test 👍🏻

MilosKozak, 2024-11-30:
result?

olorinmaia, 2024-11-30:
I haven't been able to reproduce at least

closed 2024-11-30 by MilosKozak