nightscout/AndroidAPS #3512: DynIsfTDDSens / ISF for Boluscalc is showing incorrectly on AAPSClient and presentation is confusing

olorinmaia, 2024-11-04:
In AAPS if DynISF TDD based autosens and Dynamic sensitivity is activated alot of information is showing under the autosens icon. It makes the home screen abit more stretched than necessary.

[image]

In AAPS:
[image]

In AAPSClient:
[image]

1) The number in "( )" is different in AAPSClient, aka same as profile ISF, while in AAPS it is not the same as DynISF. How is this value calculated? I can see it is used in Bolus calculator now so it is important this value is same in both AAPS and AAPSClient. I would expect it was the same value as DynISF conjures that is used in Bolus calculator now. Is there a need for a new calculated value for this?

2) The DynISF TDD based "autosens" number is missing in AAPSClient, but see 3)

3) Should we compress the information under the Sens Icon to one line? 

If DynISF is enabled, I suggest showing the DynISF vs profile ISF and move other information to a popup-window like with TBR and IOB icon when clicking on the Sensitivity arrows. This way we can add more explanation regarding Autosens %-value, DynISF TDD based sens %-value and possibly the value for bolus calc if it has to be something else than DynISF. 

If DynISF is disabled and Autosens enabled, it can be like before with no popup and show %-value of autosens.

olorinmaia, 2024-11-04:
Bolus calc in AAPS and AAPSClient:
| AAPS | AAPSClient |
|--------|--------|
| [image] | [image] |

MilosKozak, 2024-11-30:
fixed by https://github.com/nightscout/AndroidAPS/commit/1d22d801a6520694b23ded1fac54d9908469088e

MilosKozak, 2024-11-30:
implemented by https://github.com/nightscout/AndroidAPS/commit/c22e7922458acdd851b7ce9ad66f1b383be35358

closed 2024-11-30 by MilosKozak