nightscout/AndroidAPS #3513: There is no blood glucose data for selecting OTTAI

qq254344379, 2024-11-04:
I saw that I have joined ottai on September 20, but I have no data when I am using 3.3.0-dev-c, my region is China, and ottai dynamic software has been opened 小欧态动 AAP

MilosKozak, 2024-11-04:
do you use this app? https://play.google.com/store/apps/details?id=com.ottai.seas

qq254344379, 2024-11-04:
No, mine is a Chinese mobile app store installation

qq254344379, 2024-11-04:
Can you add the developers who provide the Ottai plugin to the discussion, maybe they should be able to solve it

MilosKozak, 2024-11-04:
@ottai-developer

qq254344379, 2024-11-05:
Can you put the current Ottai of the China Mobile App Store to support access to data

> @ottai-developer

qq254344379, 2024-11-05:
I still prefer to use the mainline AAPS

> > @ottai-developer

KORuL, 2025-01-16:
I use your @ottai-developer integration and application https://play.google.com/store/apps/details?id=com.ottai.seas, but the data does not go to AndroidAPS, I also rebuilt AndroidAPS and set a breakpoint and when new data appears in Ottai, your @ottai-developer application does not send a broadcast implicit intent to the system "info.nightscout.androidaps.action.OTTAI_APP" or "info.nightscout.androidaps.action.SYAI_TAG_APP"

I also made a mini application and registered a listener for these 2 events, Ottai regularly updated data from the sensor but did not send a single broadcast

MilosKozak, 2025-01-21:
https://github.com/nightscout/AndroidAPS/issues/3768

closed 2025-01-21 by MilosKozak