nightscout/AndroidAPS #3514: profile not coming back 100%

kacperpetla, 2024-11-04:
Automation sets the profile to the desired value and time, after this time the profile does not set itself to 100%
[image]
[image]

vanelsberg, 2024-11-04:
Have seen this once (after -45 mins, temp profile stil running) , unable to reproduce. 🧐

MilosKozak, 2024-11-04:
the only reason I see is when pump is not reachable. Otherwise profile is switched within 5 min

kacperpetla, 2024-11-04:
I ran tests, set the profile to 90% for 5 minutes and after 15 minutes it was -10 and nothing happened, I went back to the older version of Dev-c and the profiles change as they should, when I was building the application the servers were probably overloaded because it was laborious, today I made a new apk, installed it on my phone and set up a virtual pump, for now everything is fine, I will test for 2 days and update the master and write if everything is fine

kacperpetla, 2024-11-04:
the connection to the pump was definitely there, since it was serving smb, as you can see in the screenshots, the profile was set to 130% at 8:40. after that time it was serving smb, I turned off the profile manually at 10:36
[image]
[image]

kacperpetla, 2024-11-04:
I changed the version to dev - c, I don't know if the logs were saved, and I didn't download them then.

MilosKozak, 2024-11-04:
create new issue with logs if it happen again

closed 2024-11-04 by MilosKozak