nightscout/AndroidAPS #3522: AAPS a Garmin body battery

Tomchytil007, 2024-11-11:
Hi, I have a question about an upcoming AAPS version. We attended a training this weekend, where we were informed that connecting our Garmin watch with AAPS will be possible in the next update. This would allow the AAPS algorithm to access data like heart rate and step count, and based on this information, the algorithm could potentially adjust insulin settings accordingly.

My question is whether it will also be possible to import data on metrics like body battery or sleep quality. After a poor night’s sleep, my son is often more fatigued, and his response to insulin can differ from usual.

Milos Kozak mentioned that the key question is whether Garmin can send this additional data outside of its own app, or if only data like heart rate and step count can be shared.

Thank you.

swissalpine, 2024-11-11:
I would like to say in advance that, from my own experience, I don't consider the sleep values to be very meaningful (I repeatedly find measured values that claim poor sleep, which definitely didn't happen and vice versa), and what's more, I'm annoyed that I can't deactivate this function because it modifies my training suggestions in a bad way.
Irrespective of this, it is certainly possible to transmit these values, but the logic that would then have to be behind it in order to reinterpret these values in profile adjustments, ISF adjustments or similar is not trivial, very individual and - in my opinion - hardly mapped by algorithms.
Also, only the body battery could be transferred, but this changes from minute to minute and decreases throughout the day ...
Maybe @robertbuessow sees it differently, but I don't see it as a valid approach.
--
There is also (my) minority opinion that an increased pulse rate should not have any effect on the therapy! A high pulse rate can also be observed in stressful situations (exams, anger, etc.) that are not accompanied by exercise. In contrast to a high pulse rate due to exercise, the sensitivity is reduced in this case and resistance may be observed due to the adrenaline. The same can happen in the case of illness (increased pulse and increased resistance), here too in contrast to sport: here a high pulse leads to a higher insulin sensitivity. But at the same time a very high pulse rate (keyword: anaerobic stress) in turn leads to resistance. 
Since AAPS will not recognize why a pulse is elevated, I do not consider both pulse and body battery to be a sensible approach.

buessow, 2024-11-13:
I agree that it's rather unlikely that the body battery will help. There is a chance that it helps with detecting long term higher insulin sensitivity because of sports but I think that's rather unlikely.

tim2000s, 2024-12-30:
Based on manually observed data, the most useful information from watches in general is:

Step count 
Sleep hours
Heart rate 

I put heart rate last because, as others have said, it needs to be used in conjunction with other factors to better indicate real information. 

Step count is useful for identifying inactivity, and modifying profile accordingly, but is generally too late for notification of activity, other than indicating when activity is finished and how long to leave an adjusted profile in place for after activity. In conjunction with heart rate, these modifications can be improved. 

Finally, sleep hours rather than sleep quality data, seems to provide a more reliable indicator of likely effects the following day, however, so far the only way I've found to get that out of the Garmin app requires paying.