nightscout/AndroidAPS #3534: [Dev 3.3.0-d] URGENT - DynISF variable sesitivity used in bolus calculator

yhonatanb20, 2024-11-15:
In dev version 3.3.0-d with DynISF on, variable sesitivity is used instead of profile ISF in the bolus calculator. This may be very problematic and potentially dangerous for some people. If profile ISF worked good for me thus far in the calculator, why enforce this change?
If bug, please fix ASAP.
If feature, please make optional (like e.g. dynISF-based target change).

olorinmaia, 2024-11-16:
Also see https://github.com/nightscout/AndroidAPS/issues/3512

MilosKozak, 2024-11-18:
Primary reason for this is eliminating profile ISF at all (replaced by DynISF strengness)
In latest dev AAPS calculates average DynISF and this is used in BolusWizard. Around this value should be your profile ISF anyway.

closed 2024-11-18 by MilosKozak