nightscout/AndroidAPS #3535: [Dev 3.3.0-dev-d] Semi-tranparent dots mess up the glucose plot

yhonatanb20, 2024-11-17:
Using version 3.3.0-dev-d with xdrip-G7 as data source and average smoothing ON, some datapoints on the BG plot appear semi transparent, but in a kind of alternate way - this just complicates the graph...
[image]

yhonatanb20, 2024-11-17:
I mean, look, this looks funny, right?
[image]

parapenT1sta, 2024-11-17:
Try with the latest commit: https://github.com/nightscout/AndroidAPS/commit/4373ecdde4a0fc3aaa3b74ae1b9da6a769007a4e

yhonatanb20, 2024-11-17:
Thanks, trying

yhonatanb20, 2024-11-18:
@MilosKozak was commit [4373ecd](https://github.com/nightscout/AndroidAPS/commit/4373ecdde4a0fc3aaa3b74ae1b9da6a769007a4e) meant to fix this issue?

MilosKozak, 2024-11-18:
Transparent means recalculated gap

yhonatanb20, 2024-11-18:
So I suspected. But when using average smoothing half of the values are recalculated, this makes visualization very messy:
[image]
Knowing which values are recalculated is theoretically valuable, but practically it just makes the graph much less readable...

yhonatanb20, 2024-11-18:
Given that if I understand correctly, the white circles represent original values, maybe just bring them forward on top of always opaque BG dots?

MilosKozak, 2024-11-18:
well you should focus on why do you have so many gaps. this should be not common. for most people it's rare

MilosKozak, 2024-11-18:
because there are original values visible on top of smooted, expecting they are backfilled later

yhonatanb20, 2024-11-18:
> well you should focus on why do you have so many gaps. this should be not common. for most people it's rare

These are not gaps in the readings, this results from the smoothing algorithms. I use Xdrip + G7, and "Average Smoothing or Exponential Smoothing MUST be enabled for meaningful use of the G7 / ONE+ values." It is my understanding that this way inevitably many values will be "smoothed". Most values were recalculated by the native BYODA algorithm back on G6, right? But this did not affect visualization in AAPS because it wasn't an AAPS smoothing plugin, and I don't feel I lost valuable information there because of all points being opaque.

MilosKozak, 2024-11-18:
[image]
At time of receiving this value, previous were missing. Otherwise it would be not transparent

MilosKozak, 2024-11-20:
I never had this issue with BOYDA. Is it related only to xdrip?

yhonatanb20, 2024-11-20:
> I never had this issue with BOYDA. Is it related only to xdrip?

Definetly, I too never had it with BYODA, it happens only with xdrip (which is nearly mandatory for G7)...
The issue seemed for now to disappear ~48h after the insertion of the sensor (so that the issue coinsides with the appearance of a yellow triangle)
I will update whether this reccures with the next sensor

olorinmaia, 2024-11-20:
This is not due to BYODA or xDrip but the fact that G7 got inbuilt sender while G6 you kept the same sender for 3 month and moved it between sensors. It is likely every G7 sensor change that this will happen because the time of BG arrival is unlikely to be the same interval

yhonatanb20, 2024-11-20:
It indeed seems you are correct about the yellow triangle and it may be dispensable with G7. However it does not interfere with anything.
The semi-transparent dots seem to be a different issue, since I never got them with BYODA+G6 even when the yellow triangle was up (except the cases when there indeed was a gap in the data, which is a totally different story).

olorinmaia, 2024-11-20:
I'm pretty sure we got this with BYODA and G6 😅 But since we changed transmitter so rarely it never bothered me to much. You can be lucky and hit the same time interval when changing transmitter or g7 and not experience this issue. It's not on the second it seems.

Josch24, 2024-11-26:
I also had these Dots in Master 3.2.0.4. I use G6 with two xDrip-Version and two Anubis Transmitter. When i change from one xDrip to the other, i got this Dots. It took some Time and they are gone. But Loop is running fine.

yhonatanb20, 2024-11-28:
> I'm pretty sure we got this with BYODA and G6 😅 But since we changed transmitter so rarely it never bothered me to much. You can be lucky and hit the same time interval when changing transmitter or g7 and not experience this issue. It's not on the second it seems.

OK, I can now confirm that with the next sensor the issue **did not** recur. It's interesting that I do not seem to have hit the same interval - between the last old and first new sensor readings I had a 6-min interval, but still - no semi-transparent dots!

olorinmaia, 2024-11-28:
So no yellow warning sign? Latest Dev?

yhonatanb20, 2024-11-28:
Indeed

robertrub, 2024-12-01:
I have the same problem with G6. I use 2 Anubis which means that every 12-14 days, I get the yellow triangle and AutoISF "disconnects" partially because it can't calculate the acceleration. This lasts 36h, during the recalculation. Once the recalculated data ends, all work once more.

No smoothing in xDrip. Average smoothing in AAPS. Turning it OFF doesn't change the result :(

**It is only related to the recalculation**. If my 2 Anubis are in phase, there are never those dark/light green dots.

Look at the 2nd logs I sent in this issue
 https://github.com/nightscout/AndroidAPS/issues/3413

MilosKozak, 2024-12-05:
resolved in https://github.com/nightscout/AndroidAPS/commit/25056b52b9c314b9e2b0cc771c7b1e46a70a0304

closed 2024-12-05 by MilosKozak

yhonatanb20, 2024-12-05:
Inserted a new sensor today, issue returned. @MilosKozak please reopen the issue.

MilosKozak, 2024-12-05:
not related to https://github.com/nightscout/AndroidAPS/issues/3413 , https://github.com/nightscout/AndroidAPS/issues/3197