nightscout/AndroidAPS #3537: [Feature request] Option to choose different insulin profile when using "do not bolus, record only"

rbenma, 2024-11-17:
I'm currently using Novorapid in my pump. For high carb meals or when I forget to prebolus, I use a pen with fiasp. I then record the bolus given via pen in the insulin tab with "do not bolus, record only". However, since I use fiasp in the pen, the loop uses the Novorapid insulin profile, which I guess leads to an inaccurate insulin action expectation for the loop. If you could select a different insulin action profile for such manual pen bolus, the loop would receive more accurate information on the insulin activity. Not sure this is relevant for a lot of people or if it makes a huge difference in my case as they should still have more or less the same DIA. Thanks for your consideration!

Philoul, 2024-11-25:
A very long time ago 😉 I opened this topic #269 on insulin management...

My idea was to be able to manage our own list of insulin like we manage our own list of profile, within one single "insulin plugin", where we can create and name (using 3 or more templates) one or several insulins we often use (in pump, in pen, ...) with all parameters (peak, DIA, and maybe more 😉), and select within profile one of these insulin.

Even if for most users only one insulin is used, this allow selection for external injection by pen to select a different one...

olorinmaia, 2026-09-23:
Will be possible in AAPS 4.0.0, currently dev

closed 2026-09-23 by olorinmaia