nightscout/AndroidAPS #3538: Activating user actions button should not use this big screen-wide button on overview

vanelsberg, 2024-11-18:
**AAPS 3.3 dev**

**Suggestion**:

Automation user action uses a big screen-wide button on AAPS overview.
Imho the button is to ease to hit by accident and uses far too much valuable screen space in overview (more than 10%)? 

I can see several options here?
- Make the button much smaller
- Moving user automation buttons to the "Actions" tab, or even a separate "Automations tab"?
- Moving user automation buttons to a new/separate "Automations" tab?

MilosKozak, 2024-11-18:
Hmmm there are always different kind of users. Like you and people for whom has the button same importance like Calculator

vanelsberg, 2024-11-18:
Agree. But it's a bit similar to the Quick Wizard buttons. I would never use them from the phone. On watch however imho they are much more useful?

MilosKozak, 2024-11-20:
https://github.com/nightscout/AndroidAPS/commit/df58fe4fa1a04924a479a044c26f5babcd8adda9
Let call this a precedent how to manage options suitable for small amount of users

vanelsberg, 2024-11-20:
> [df58fe4](https://github.com/nightscout/AndroidAPS/commit/df58fe4fa1a04924a479a044c26f5babcd8adda9) Let call this a precedent how to manage options suitable for small amount of users

Great! Perfect solution for me. Love it!
Tnx!