nightscout/AndroidAPS #3550: [Dev] Preferences for Pump and Local alerts appear in places they don't belong

olorinmaia, 2024-11-23:
Navigate to Home Screen -> Preferences ->  Overview -> Default Temp-Targets or Fill/Prime or many other preferences via Home Screen -> Preferences.

"Pump" and "Local alert" appear at the bottom for some reason. I've been trying to fix this myself but no luck so far.

[image]

Philoul, 2024-11-25:
I confirm this issue, very strange...

It's for all preferences that have 2 level of "standard" sub menu.
Within second level of submenu (opened in a dedicated page), it's like pump/advanced and alert/advanced are not removed or hidden and remained visible...

MilosKozak, 2024-11-27:
fixed by https://github.com/nightscout/AndroidAPS/commit/97860cf0cf1b35e151f12174ba0bba29e077f750

closed 2024-11-27 by MilosKozak