nightscout/AndroidAPS #3555: [Dev 3.3.0-dev-d] Cannot do "Maintenance/Import Settings" and "Wear/Load Watchface" because AAPS crashes immediatelly.

reijoahola-nightscout, 2024-11-25:
Phone: Samsung Galaxy S21 5G,  ndroid 14, One UI 6.1, latest updates donr

The phone was earlie used as main AAPS with master 3.2.0.4 and worked without this problem.
I took this phone as a test phone because of Android 14 and tried to install the latest dev.
I could not complete a clean install with the latest dev but it was possible with some earlier dev.
Then I installed Preferences from my 3.2.0.4 master AAPS.
Then I updated into the lates dev and seemed to go smoothly.

But the I found at leas two problems
- Custom watchfaces cannot be selected -> AAPS crash
- Settings cannot be impoeted -> AAPS crash

AAPS directory is normally selected and access allowed as requested.
Exporting settings is however possible and creates a new preference file.

I send a log file with AAPS and attach number of this Issue.

AAPS 3.3.0-dev-d dated 984167245c-2024.11.25 
[image]

MilosKozak, 2024-11-25:
try latest dev

reijoahola-nightscout, 2024-11-26:
Installed the latest dev (f3d464036) from yesterday evening.
- now it did not crash but gave an error about folder (do not remember exact error text)

Created .../Internal storage/AAPS again and waited when AAPS app created the subfolders
- Maintenance/Import Settings still gave an error about folder
- ... but Wear/Load Watchface opened and could change Custom watchface
- ... but only once and when trying again watchafaces folder, it gave the same error about folder

Next I created AAPS again but now under Documents folder (.../Internal Storage/Documents/AAPS)
- now both folders seems to open normally and I tried it several time
- also Export Settings still work
- ... but if I try to copy some new custom watchface zip file into exports folder, it again gives that error about folder

Now I have in export folder only those files that AAPS created and it seems to work.
Is there some rule that AAPS folder should not be directly under Internal Storage and should be created for example Under Documents folder ?

I continue tests with dev installation and try to backup enough TDD data from NS so that I can use Auto ISF.

MilosKozak, 2024-11-27:
@Philoul can you replicate?

Philoul, 2024-11-27:
No, I never got crash during CWF file selection.. But file RW access works fine on my config or emulator when I updated my config. After AAPS folder selection, I approved Android access request for AAPS and it worked (new files externally added from computer in exports folder are visible, ...)
I managed (I think) all potential crashes due to wrong zip file format, like wrong content, wrong zip, so only "good files" are visible and all the others are hidden withoit crash.
Here I think crash (if remaining after your update) is probably more due to issue during access to folder content (list of files) or when we try to access to file binary data 🤔, but it should not be limited to Wear plugin...

reijoahola-nightscout, 2024-11-27:
Crashes were with earlier dev versions. After AAPS 3.3.0-dev-d dated 2024.11.25 there are no crashes but
during certain situations (explained earlier), it gives an error "Error accessing file system ....".

When I have now selected AAPS file as ../Internal Storage/Documents/AAPS, Import Settings does not cause problems.
Also Load Watchfaces does not cause error message if I do not add custom watchface.

Note, AAPS master worked in this phone without these problems.
Also with some older dev version I could install (clean install) and load Preferences file but
problems are with these newer AAPS dev versions that use user selectable AAPS directory (I believe so but not sure).

Philoul, 2024-11-27:
The strange issue (if I understood you correctly) is that you can select all default CWF created by AAPS (from assets), but when you manually add another zip file within exports folder (could be with phone file manager, or with computer explorer), then new files are not visible 🤔 (looks like authorization is attached to each file and not folder content).

I cannot reproduce this error, all zip files put within exports folder, if they have correct content, are visible and can be selected... (when I develop or tune a new CWF, I spend my time to update "exports folder" from my computer, with new files, updated files... and everything is immediatly visible within Wear plugin).

reijoahola-nightscout, 2024-11-27:
Yes, I downloaded "Blue Ring" watchface by using Wear/More Watchfaces. Then moved zip file into ...Internal storage/Documents/AAPS/exports. After this Wear/Load Watchface gives an error "Error accessing filesystem.....".
I tested earlie with some other custom watchface and same result so probably it is not that specific watchface that cause the error but something related access rights etc. for these added files. BTW, how to see access rights for files and folders in Android ?

Philoul, 2024-11-28:
@reijoahola-nightscout Do you have same issue with preferences ? (if you manually copy/Paste an external json preference file from another phone within preferences folder, can you see and select them or can you only select preferences json files saved directly by running version of AAPS on phone ?)

reijoahola-nightscout, 2024-11-28:
"Do you have same issue with preferences": I think I did not test with such way very much but ptrobably it also caused a problem.

But I have now new information about the problem and better put this into hold for some time.
I started to believe vthat my phone (file system) was somehow corrupted and decided to do total reset (factory reset).
I have now restored the installation and first impression is that now it works normally
- AAPS directory can be .../Internal Storage/AAPS
- can Import settings normally
- can open Custom Watchfacesa folder

I have not yet stressed the system much. I tell later if it still cause problems.

reijoahola-nightscout, 2024-11-29:
After more extensive testing I am sure that now it works and the root cause was some corruption in my phone Android file system. I think this issue can now be closed.

closed 2024-11-29 by MilosKozak