nightscout/AndroidAPS #3560: Intergrate Xdrip Sync method to AAPSClient, without the need for xDrip.

staticx, 2024-11-27:
Hey, 

Nowadays, we have the alternative of NOT using, nor building Nightscout - we can just use xDrip Sync to our followers and maitain history and data in Tidepool. 
The thing is, when you're a follower, you must use xDrip to see BG... So basically if I want to use AAPSClient, I got to install xDrip too. 
Is would be awesome to have xDrip Sync as BG data in Config Builder of AAPSClient, this way we won't need to install xDrip at all....

thanks !

andyrozman, 2024-11-27:
If you use "xDrip Follow", then you have to have xDrip installed on both phones... We can't integrate xDrip into AAPS(Client) just so that you don't need to install 2nd application. Follow functionality of xDrip is not that simple and it shouldn't be in AAPS. Maybe talk to people in xDrip if they can create more simple xDripClient app, but you will probably get similar answer from them.

MilosKozak, 2024-11-27:
and more: If I remember well xDrip is using Google services for sync which means you have to use the same Google account on all devices which is not practical from many reasons

closed 2024-11-27 by MilosKozak

olorinmaia, 2024-11-30:
@MilosKozak xDrip doesn't use Google services anymore as they shut down the service, they have setup their own service now at Bluejay project. All you need is to use the same key between master phone and follower phones and BG + data from AAPS will sync between the xDrip's flawlessly. Remote snoozing of alerts is also a neat feature :) And there is no need for the same google account anymore.
https://navid200.github.io/xDrip/docs/xDripSync.html

To be able to use AAPSClient properly you need NS anyway.

MilosKozak, 2024-11-30:
thanks for clarifying 👍