nightscout/AndroidAPS #3563: [Improvement] Improve UX for Config Builder by only showing selected options

olorinmaia, 2024-11-27:
The Config builder is starting to get very long with the growing pump/cgm support. Should we considering adding some sort of minimizing feature? That way all non-selected options can be automatically hidden, and only show the ones that you have selected, and add a button so you can expand the list to show all options. 

Thoughts or possibly other ideas how to improve Config Builder?
Illustration of how it could be done:
[image]

How the long the list is actually today :)

[image]

olorinmaia, 2024-12-02:
I like it :) 

Do you have a branch for this to be tested ? or just early stages trying to visualize it?

kenzo44, 2024-12-02:
This is just a quick mockup in figma. 

It will behave similar to the preferences page - Users can see the headings once they open the config builder and choose which one they would like to expand.

Initially, I was thinking of doing it like the android settings page. Instead of dropdowns, having separate pages open for each category.

[image]

We need to update the preferences page as well - We can expand headings but not collapse them.

olorinmaia, 2024-12-02:
Great idea about improving preferences also, I was looking at how to remove the word "advanced" from every block but it seemed to origin from a read-only file so that kind of stopped me as of now ;)

MilosKozak, 2024-12-03:
https://github.com/nightscout/AndroidAPS/commit/37b5386875c95861da76288b19b5d0314fe1b8b9

closed 2024-12-03 by MilosKozak