nightscout/AndroidAPS #3568: AverageDynISF for Bolus Calculator to follow Profile Switch % - strength

olorinmaia, 2024-11-30:
In 3.3 Average DynISF for the last 24hours will be introduced that will be used in Bolus Calc ISF instead of Profile ISF if Dynamic Sensitivity is enabled.

If you set a profile switch to 80% or 120% this average ISF value is not modified like regular Profile ISF when Dynamic sensitiviy is disabled. My concern is that ISF will be to weak or strong given the situation and we should be able to weaken or strengthen this number with profile switch %-strength in my opinion.

From the code the way I read it the DynISF average value will be calculated from the last 24h.

```
    override fun getAverageIsfMgdl(timestamp: Long, caller: String): Double? {
        var count = 0
        var sum = 0.0
        val start = timestamp - T.hours(24).msecs()
        dynIsfCache.forEach { key, value ->
            if (key in start..timestamp) {
                count++
                sum += value
            }
        }
        val sensitivity = if (count == 0) null else sum / count
        aapsLogger.debug(LTag.APS, "getAverageIsfMgdl() $sensitivity from $count values ${dateUtil.dateAndTimeAndSecondsString(timestamp)} $caller")
        return sensitivity
    }
```

koelewij, 2024-11-30:
This is a tough one in my opinion. Though I am not using DynISF anymore for a long time, the core of DynISF is that it ignores your profile ISF completely and relies solely upon its own calculated ISF based on TDD and BG. If your sensitivity changes and you change your profile % your TDD will change and so also your resulting ISF. So if the ISF used in the calculator is the value based on the ISF from the last 24hrs, I would say that value is correct then. 
I also wonder what you would like to use then. The same value compensated by profile %?  That might work in case you just that % but after 24hrs your ISF has already changed accordingly and then compensating it is too much in my opinion. Did you have issues with the current results?

olorinmaia, 2024-12-01:
I dont use dev on "Live" so I can't really say for sure if this is going to be an issue, but i want input from others about what is the best way forward here. :)

The most common use case for us to strengthen profile is when kid is falling to sleep (to 130%) and when exercising reducing to 80%. This is kind of resistance/sensitivity that appears in an instant, and dynisf won't be able to adjust to that quickly enough. Bolus calc is only applied maybe once in these use cases before exercise and perhaps once after going to sleep to get a kickstart before growth hormons kick in :)

MilosKozak, 2024-12-01:
If the profile stay on 130% for a longer time, ISF if already adapted to this profile and adding +30% to the profile would multiply the effect

vanelsberg, 2024-12-01:
When ISF is calculated - so independent from profile value - to me it does not make sense to change it when profile changes - even if it is only by %?

olorinmaia, 2024-12-01:
Thanks for input. I'm closing this now and I agree it's best to start like it is now.

closed 2024-12-01 by olorinmaia