nightscout/AndroidAPS #3571: [Dev-E] NSV3 (and possibly V1) is making duplicate entries to OpenAPS-log in AAPSClient and NS

olorinmaia, 2024-12-01:
AAPS / Client 3.3.0-dev-e
Build: cab69e7cae-2024.11.30
Nightscout version: 15.0.2
Engineering mode enabled 

In AAPSClient messages are duplicate and identical, in NS the information is abit different, one with mmol/l and other with mg/dl and some temp basal info.

Logs:
AAPS:[AndroidAPS_LOG_1733057278672.log.zip](https://github.com/user-attachments/files/17968944/AndroidAPS_LOG_1733057278672.log.zip)

AAPSClient: [AndroidAPS_LOG_1733057260440.log.zip](https://github.com/user-attachments/files/17968942/AndroidAPS_LOG_1733057260440.log.zip)

In AAPSClient messages are identical:
[image]

In NS there is some difference:
[image]

emilisev, 2024-12-06:
Same issue for me on beta1

olorinmaia, 2024-12-14:
Latest NS Dev with merged https://github.com/nightscout/cgm-remote-monitor/pull/8234
Fixes issue with showing mg/dl but still duplicate openaps entry. Here with identical output
[AndroidAPS_LOG_1734219330736.log.zip](https://github.com/user-attachments/files/18138149/AndroidAPS_LOG_1734219330736.log.zip)

[image]
[image]

olorinmaia, 2024-12-17:
Relevant comment here: https://github.com/nightscout/AndroidAPS/issues/3620#issuecomment-2549008008

bewest, 2024-12-18:
If possible, I recommend delaying sending a devicestatus until suggested and enacted can be reported as a single record.  This is how things worked with openaps.

MilosKozak, 2024-12-29:
from my point of view it's correct. Status is generated on more places (not only after enact) for example for cases like open loop, failed enact etc.
I added logging anyway so you can see the reason https://github.com/nightscout/AndroidAPS/commit/5e706888e0982b8e5a626a3c8a3afe27217523d9

closed 2024-12-29 by MilosKozak