nightscout/AndroidAPS #3578: [3.3.0-dev-e] Fallback to profile sensitivity notification

WesSec, 2024-12-03:
Hi, i recently built the new dev version (6201b8050c-2024.12.02), imported my historic data from NS (making historic TDD available).

Once a day i receive the notification that there is not enough TDD data and it falls back to profile sensitivity

[image]
TDD is availble:
[image]

In discord i briefly announced this, where it was a known issue using the AutoISF plugin. I am using the SMB plugin.

Here is a snippet of the logs containing the message, i could not observe any surrounding rows causing this behavior

```
16:47:11.850 [RxCachedThreadScheduler-104] D/APS: [ProfilerImpl.log():14]: >>> getIsfMgdl() GLUC null 02/12/2024 16:47:11 XdripPlugin <<< executed in 4 milliseconds
16:47:11.850 [RxCachedThreadScheduler-30] D/NOTIFICATION: [NotificationStore.add():62]: Notification received: Fallback to profile sensitivity. Not enough TDD data.
16:47:11.863 [RxCachedThreadScheduler-59] D/APS: [ProfilerImpl.log():14]: >>> getIsfMgdl() GLUC null 02/12/2024 16:47:11 XdripPlugin <<< executed in 16 milliseconds
16:47:11.877 [RxCachedThreadScheduler-30] D/NOTIFICATION: [NotificationStore.add():62]: Notification received: Fallback to profile sensitivity. Not enough TDD data.
16:47:11.885 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():249]: ► UPL Start
16:47:11.888 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():249]: ► ADD entries Sent PairGlucoseValue <i>{"date":1733151124114,"device":"G7","direction":"FLAT","filtered":0.0,"isReadOnly":false,"isValid":true,"sgv":170.0,"unfiltered":0.0,"units":"MG_DL","utcOffset":60}</i> 32121/32132
16:47:12.810 [RxCachedThreadScheduler-75] D/AUTOMATION: [AutomationPlugin$onStart$12.accept():190]: Grabbed new BT event: EventBTChange[deviceAddress=CD:F0:C3:2D:48:5B,deviceName=DXCMxn,state=DISCONNECT]
16:47:12.815 [RxCachedThreadScheduler-75] D/AUTOMATION: [AutomationPlugin.processActions$automation_fullRelease():266]: processActions
16:47:15.550 [main] D/CORE: [DataReceiver.onReceive():38]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 10; com.eveningoutpost.dexdrip.Extras.BgSlope => 1.3333333333333333E-5; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2124091707 dfcbe80-2024.09.17; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Display.Units => mmol; com.eveningoutpost.dexdrip.Extras.Time => 1733154123635; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => null; com.eveningoutpost.dexdrip.Extras.SensorStartedAt => 1732563841577; com.eveningoutpost.dexdrip.Extras.Noise => -9999.0; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 184.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => G6 Native / G5 Native; com.eveningoutpost.dexdrip.Extras.SourceInfo => G7::Backfill; com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus =>   ; }Bundle
16:47:15.605 [DefaultDispatcher-worker-5] D/BGSOURCE: [XdripSourcePlugin$XdripSourceWorker.doWorkAndLog():103]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=10, com.eveningoutpost.dexdrip.Extras.BgSlope=1.3333333333333333E-5, com.eveningoutpost.dexdrip.Extras.VersionInfo=2124091707 dfcbe80-2024.09.17, com.eveningoutpost.dexdrip.Extras.Raw=0.0, com.eveningoutpost.dexdrip.Extras.Display.Units=mmol, com.eveningoutpost.dexdrip.Extras.Time=1733154123635, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=-1, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.SensorStartedAt=1732563841577, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=-1, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=184.0, com.eveningoutpost.dexdrip.Extras.SourceDesc=G6 Native / G5 Native, com.eveningoutpost.dexdrip.Extras.SourceInfo=G7::Backfill, com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus=  }]
16:47:15.609 [RxCachedThreadScheduler-192] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():62]: Firing EventNewBG GlucoseValue(id=32133, version=0, dateCreated=1733154435607, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1733154123635, utcOffset=3600000, raw=0.0, value=184.0, trendArrow=FLAT, noise=null, sourceSensor=UNKNOWN)
16:47:15.610 [RxCachedThreadScheduler-105] I/GARMIN: [GarminPlugin.onNewBloodGlucose():181]: onNewBloodGlucose Mon Dec 02 16:42:03 GMT+01:00 2024
16:47:15.611 [RxCachedThreadScheduler-192] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():67]: Firing EventNewHistoryData 1733154123635 GlucoseValue(id=32133, version=0, dateCreated=1733154435607, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1733154123635, utcOffset=3600000, raw=0.0, value=184.0, trendArrow=FLAT, noise=null, sourceSensor=UNKNOWN)
16:47:15.612 [RxCachedThreadScheduler-192] D/DATABASE: [PersistenceLayerImpl$insertCgmSourceData$4.apply():585]: Inserted GlucoseValue from Xdrip GlucoseValue(id=32133, version=0, dateCreated=1733154435607, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1733154123635, utcOffset=3600000, raw=0.0, value=184.0, trendArrow=FLAT, noise=null, sourceSensor=UNKNOWN)
16:47:15.613 [DefaultDispatcher-worker-5] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.plugins.source.XdripSourcePlugin$XdripSourceWorker Data {}
16:47:15.615 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():249]: ● RUN Already running NEW_DATA
16:47:15.623 [RxCachedThreadScheduler-59] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():38]: oldData
16:47:15.625 [RxCachedThreadScheduler-104] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():38]: oldData
**16:47:15.625 [RxCachedThreadScheduler-30] D/NOTIFICATION: [NotificationStore.add():62]: Notification received: Fallback to profile sensitivity. Not enough TDD data.**
16:47:15.626 [RxCachedThreadScheduler-59] D/APS: [ProfilerImpl.log():14]: >>> getIsfMgdl() GLUC null 02/12/2024 16:47:15 XdripPlugin <<< executed in 4 milliseconds
16:47:15.626 [RxCachedThreadScheduler-30] D/NOTIFICATION: [NotificationStore.add():62]: Notification received: Fallback to profile sensitivity. Not enough TDD data.
16:47:15.626 [RxCachedThreadScheduler-104] D/APS: [ProfilerImpl.log():14]: >>> getIsfMgdl() GLUC null 02/12/2024 16:47:15 XdripPlugin <<< executed in 2 milliseconds
16:47:20.620 [pool-9-thread-1] D/WORKER: [CalculationWorkflowImpl.stopCalculation():47]: Stopping calculation thread: onEventNewHistoryData
16:47:20.641 [pool-9-thread-1] D/WORKER: [CalculationWorkflowImpl.stopCalculation():52]: Calculation thread stopped: onEventNewHistoryData
16:47:20.645 [pool-9-thread-1] D/WORKER: [CalculationWorkflowImpl.runCalculation():64]: Starting calculation worker: EventNewBG to 02/12/2024 16:47:20
```

Some context
- Samsung S21FE (android 14)
- Fresh install, imported historic data via NSclientv3

interval is around 24 hours i think, on main screen the sens is working as expected

MilosKozak, 2024-12-03:
I improved the messaage in dev

WesSec, 2024-12-03:
Thanks! looking at your explanation it's likely that data is being backfilled by xdrip because of a missed bg due to out-of-range sensor or something, which can happen on various occasions, it would be a nice improvement if the notification can be suppressed, or not being shown in this instance.