nightscout/AndroidAPS #3585: [beta1] TBR not showing in AAPSClient with NSv1

emilisev, 2024-12-06:
AAPS and AAPSClient both on beta1, both using NS v1. Settings imported from 3.2 on both sides. I don't see the TBR on AAPSClient. I see them all right on NS web UI, so they get uploaded.

[AAPSClient log](https://github.com/user-attachments/files/18042466/AndroidAPS_LOG_1733511287304.log.zip)
[image]

[image]

MilosKozak, 2024-12-07:
was the data uploaded to NS by 3.3?
pm me access to you NS to be able see the data

emilisev, 2024-12-07:
No, the data is uploaded with v1 as well. Credentials sent in discord, thanks.

MilosKozak, 2024-12-07:
resolved in https://github.com/nightscout/AndroidAPS/commit/0247d3b198fc572889122e0eb3c75acceb26028e

closed 2024-12-07 by MilosKozak