nightscout/AndroidAPS #3586: [3.3 beta1] Change settings orer in SMB section

emilisev, 2024-12-07:
Screenshot attached shows in red the settings visible only when SMB enabled. UAM is not part of them (always visible) and shows in the middle. Settings should be reordered so that all settings related to SMB are next to each other.
[image]

olorinmaia, 2024-12-07:
I personally don't mind the way it is ordered now. But is it better if Enable UAM is to be moved so it is above Minimum carbs required?

As far as I know UAM predictions work altho SMB is disabled, that's why it's left out of the dynamic view when smb is disabled/enabled.

If no other suggestions I can make a PR to adjust this.

emilisev, 2024-12-07:
What bugs me is that when you check/uncheck "SMB enabled", these settings show/hide. I find it confusing for new users that the settings that show when enabling another are not just below it.
But there is the issue of "max minutes of basal for smb + uam" that only shows when both are activated, so it should go below UAM as well ? Then it gets separated from "max minutes of basal for smb", which doesn't make sense...
For my brain, the best logical order is this one : 
[image]

WesSec, 2024-12-07:
I agree about the confusing UI behavior with SMB settings. While exploring this further, I noticed a similar UX issue: the "Use Autosens feature" disappears when "Enable dynamic sensitivity" is toggled. This made me wonder whether Autosens was disabled or just hidden from the UI.

This pattern of settings appearing/disappearing based on other toggles can be disorienting, especially for new users. While your suggested logical order makes sense, we might want to consider broader UI improvements like:
- Using indentation or visual hierarchy to show relationships between settings
- Implementing a submenu structure for related controls
- Don't hide but grey out settings when not applicable.

The "max minutes of basal" setting is a perfect example of this challenge - it needs to be visually connected to the SMB Enabled setting while maintaining a logical flow.

emilisev, 2024-12-07:
Agreed. I feel like settings' visibility depending on another setting is new, or at least getting more widely used, in AAPS (same behavior used in the new "Simple mode"). I think this is a good way to make more clear which settings are actually used, and simplify usage for new users who don't have much enabled, but also agree that this could be improved visually.
I think the grayed out option + explanation on why this setting is currently disabled / how to make it available provides the most information, but probably doesn't go in the way of simplification...

olorinmaia, 2024-12-07:
I agree the settings everywhere in AAPS could be improved. I also think it's useful to hide settings that have no effect when certain other settings are enabled, good example is Enable SMB Always or Autosens when DynISF is enabled because it simply has no effect.. Question if this is best by hiding or greying out settings I'm not sure what is the best approach. 

I like the idea by grouping up the settings more, so one could see who belongs to eachother. Problem is there are so many settings, and if all are showing all the time that may not be working at all that will be even more confusing for a not experienced user.

I made a small PR to address main feedback and fix another issue. :) And for sure this could be further improved now that all settings are moved to kotlin.

MilosKozak, 2024-12-07:
https://github.com/nightscout/AndroidAPS/commit/cf9af90f7a5bf4c54870444d8f18d362f2cf6263

closed 2024-12-07 by MilosKozak