nightscout/AndroidAPS #3587: 3.3.0 Beta 1 AAPS Sync/Websockets

elgrupo123, 2024-12-07:
Wenn Websockets aktiviert sind, bekommt AAPS Client keine Daten. Sind die Websockets auf allen Geräten ausgeschaltet, funktioniert der Sync. Ist es so gewollt??
Das Menü "Synchronisierung" in AAPS Client lässt sich nicht öffnen.

MilosKozak, 2024-12-07:
must be related to your connectivity as the communication module is the same on master and client

elgrupo123, 2024-12-07:
Also ich muss auf beiden Follower Geräten die Websockets ausschalten, um die Daten zu bekommen, Master Websockets sind aktiviert. Wenn ich AAPS Clients ältere Versionen nehme, können die Websockets aktiviert bleiben und die Daten kommen an.

MilosKozak, 2024-12-07:
provide screenshots and logs if you need more help

closed 2024-12-11 by MilosKozak

elgrupo123, 2024-12-17:
Das Problem besteht selbst mit der DEV Version. Nach einiger Zeit werden NUR die BZ Werte nicht synchronisiert, alle anderen Daten kommen an.  Heute Morgen ca. 6Uhr tratt der Fehler wieder auf. 
Mit der Master Version (3.2.0.4) gibt es das Problem nicht
[AndroidAPS._2024-12-17_00-00-19_.0.txt](https://github.com/user-attachments/files/18167163/AndroidAPS._2024-12-17_00-00-19_.0.txt)

elgrupo123, 2024-12-17:
> Stellen Sie Screenshots und Protokolle zur Verfügung, wenn Sie weitere Hilfe benötigen

Das Problem besteht selbst mit der DEV-Version. Nach einiger Zeit werden NUR die BZ-Werte nicht synchronisiert, alle anderen Daten kommen an. Heute Morgen ca. 6Uhr tritt der Fehler wieder auf.
Mit der Master Version (3.2.0.4) gibt es das Problem nicht.

[AndroidAPS._2024-12-17_00-00-19_.0.txt](https://github.com/user-attachments/files/18167252/AndroidAPS._2024-12-17_00-00-19_.0.txt)