nightscout/AndroidAPS #3590: [3.3 beta1] Feedback on unattended exports

emilisev, 2024-12-08:
Feature working well so far, just some feedback on strings.
Here's what I did on this screenshot : 
* 16:52 : unattended export attempt through automation
* 16:53 : manual export
* 16:55 : unattended export through automation

My feedback : 
* 16:52 : the text seems so long that the checkbox to delete the entry does not show
* 16:52+16:55 : strings improvements : empty parenthesis, nothing before colon
* 16:55 : the source string seems to be different than the manual export one. Maybe it's the same wording in english, but the resulting possible discrepancies in translations makes it confusing.

[image]

emilisev, 2024-12-09:
> the password for automatic export expire once a month and you have to re-enter it within 1 week to continue doing exports. the reason is for this is that if you don't remeber the password, the export is useless. 

Yes, no problem with that, my feedback was only about UI. Here is the screenshot in english showing the same issues : 

[image]

MilosKozak, 2024-12-09:
fixed by https://github.com/nightscout/AndroidAPS/commit/9c041ef1480a0693434e46de868b626616e98d74

closed 2024-12-09 by MilosKozak