nightscout/AndroidAPS #3591: Wear menu in AndroidAPS is not fully shown on coverscreen of Samsung Z Flip series

koelewij, 2024-12-08:
Reporting bugs
--------------
In AAPS V3.2.0.4 and V3.3.0 beta-1:
Where all other menus/tabs in AndroidAPS are scrollable, the Wear tab is not showing this behaviour. 
In portrait mode on a normal phone screen, this is no issue, but on the coverscreen of the Samsung Z Flip series, this causes the bottom of the menu to become inaccessible.
[image]
It is also showing this behaviour on the main phone screen when it is in landscape mode. 
According to @Philoul the behaviour of this tab is different to be able to fill the bottom of the screen with the preview of the active watch face regardless of the screen resolution.

Though this is not a high priority issue (phone can be opened or placed in portrait mode to be able to see the full screen), a more flexible solution would be welcomed. Also taking into account the fact that in the future the number of buttons on the Wear screen could increase and would require adjustments also. 

This issue is mainly recorded on request of @Philoul to keep this on the list.

olorinmaia, 2024-12-09:
Could you try https://github.com/nightscout/AndroidAPS/pull/3592 ?

Philoul, 2024-12-09:
@koelewij  If you confirm it's fixed with @olorinmaia 's commit, you can close this issue...

koelewij, 2024-12-09:
@Philoul PR3592 by @olorinmaia has fixed this issue.
I will close this issue. Also because @MilosKozak already merged it into dev.

closed 2024-12-09 by koelewij