nightscout/AndroidAPS #3594: AiDEX CGM plugin does not work and is no longer relevant - Should be removed in next version

Stealthwick, 2024-12-09:
AiDEX CGM / GlucoRX plugin has never actually worked to my knowledge (however it was possible to use AiDEX CGM via xDrip).

AiDEX CGM has now been replaced by the much improved and more consistent LinX (or AiDEX X in China). Integration is possible via companion app mode in xDrip. The AAPS AiDEX option isn't needed and has never worked anyway.

andyrozman, 2024-12-09:
Aidex CGMS plugin was receiving data from (initial and for us modified) Aidex App. Initial version worked as expected. Same as Aidex Xdrip plugin. 
If they changed something and made plugin not work, I don't know. I was communicating with them (at that time) and they integrated broadcast functionality which is the way we were receiving data on both Apps (AAPS and Xdrip).

MilosKozak, 2024-12-09:
[image]
No sign of original AidexPlugin.
I'll remove it

closed 2024-12-09 by MilosKozak