nightscout/AndroidAPS #3595: 3.3.0-beta1 Turn on "Enable dynamic sensitivity"  and excuting Loop then app stop.

9mjn9mj, 2024-12-09:
-version: 3.3.0-beta1   / android 12 (one UI 4.1) 

when turn on "Enable dynamic sensitivity" 

in Loop  menu, pull down to excute then it stop the app. 

https://github.com/user-attachments/assets/6a1b1638-a7a3-4145-b289-34f4c3b26ae7

log file:
[AndroidAPS_LOG_1733737020523.log.zip](https://github.com/user-attachments/files/18059503/AndroidAPS_LOG_1733737020523.log.zip)

(clear the log, and then  reproduce the situation. and save log)

olorinmaia, 2024-12-09:
Confirming, adding logs with latest dev. See around 14:30-14:31
AAPS 3.3.0-dev-e
Build: 5791364148-2024.12.09
[AndroidAPS_LOG_1733751110948.log.zip](https://github.com/user-attachments/files/18062170/AndroidAPS_LOG_1733751110948.log.zip)

MilosKozak, 2024-12-09:
I see in log that AAPS crashed but the reason is not there and I was not able to replicate it.
Do you have config that crashes regulary to ba able replicate it?

MilosKozak, 2024-12-09:
if I import it on fresh phonewill it crash?

olorinmaia, 2024-12-09:
Will probably need to activate engineering mode and get enough bg in from random bg to activate loop!

MilosKozak, 2024-12-09:
fixed by https://github.com/nightscout/AndroidAPS/commit/d511e9e1fc806a3bd3497233b1e16b0045861d3c

closed 2024-12-09 by MilosKozak