nightscout/AndroidAPS #3597: [Dev] [NSv3] TBR sync after short period of offline sync wrong on AAPSClient causing wrong IOB

olorinmaia, 2024-12-09:
After a short period of no connectivity there seem to be a small issue with TBR syncing correctly causing wrong IOB in AAPSClient vs AAPS.

Logs:
AAPS: Not able to find logs from that time period.. Shouldn't they be located in \Android\data\info.nightscout.androidaps\files ? Don't seem to write in the folder set by setting in Maintenance.

AAPSClient: 
[AndroidAPS_LOG_1733776611326.log.zip](https://github.com/user-attachments/files/18067808/AndroidAPS_LOG_1733776611326.log.zip)

| AAPS | AAPSClient |
|--------|--------|
| [image]| [image]|

MilosKozak, 2024-12-10:
is there a step by step guide how to replicate it?

olorinmaia, 2024-12-10:
I tried yesterday but couldn't.. I'll make another attempt later today. It could be a rare case.

olorinmaia, 2024-12-10:
Can't reproduce. Will open again if I manage.

closed 2024-12-10 by olorinmaia