nightscout/AndroidAPS #3606: Temporary profile percentage for a duration set does not end.

vanelsberg, 2024-12-11:
**AndroidAPS 3.3.0-dev-e**
**Build:** 859764d2ce4fc3359c4fe4e243a89466fc5aa996
**Flavor:** Fullrelease

**Problem description:**

Temporary profile percentage for the duration set does not end.
Temporary target was manually to 80% for a duration of 180 minutes at 02:09hrs.
In overview at 0918hrs it is still active, showing negative time of 9 minutes. At that time however, 80% profile is still active and does not change/progress.

**Additional information**
Problem looks similar to issue https://github.com/nightscout/AndroidAPS/issues/2538

_Logfiles_
[AndroidAPS_LOG_1733905150983.log.zip](https://github.com/user-attachments/files/18092575/AndroidAPS_LOG_1733905150983.log.zip)

_Screenshots:_
[image]
[image]

vanelsberg, 2024-12-11:
Ah yes - realize there is much more logs then just latest 3 :-|

All logs from today:
[logs.zip](https://github.com/user-attachments/files/18094287/logs.zip)

vanelsberg, 2024-12-11:
> negative time can easy happen between end of profile with duration and enacting the change by pump.

Yes, but I think this is different: At this moment AAPS is **still running op 80% profile** (12 hours after setting 80% for 180 minutes. DASH history confirms it did not cancle.... (this should also see that in the logs?)

Have to manually reset bij either set a profile of 100% or delete the temp profile% from treatments...

MilosKozak, 2024-12-11:
AAPS didn't schedule "KeepAlive". this is happening sometimes on some phones and as it's google's library I cannot affect it.
Restart of AAPS probably helps.
You may search logs for "KeepAlive" to be sure it works

MilosKozak, 2024-12-11:
I don't see it it newest logs as well

closed 2024-12-11 by MilosKozak

vanelsberg, 2024-12-14:
Hmmm.... 
Will monitor. And have a closer look at the logs when problem repeats on current dev