nightscout/AndroidAPS #3607: Layout degradation at AAPS portal overview: unexpected empty space

vanelsberg, 2024-12-11:
**AndroidAPS 3.3.0-dev-e**
**Build:** https://github.com/nightscout/AndroidAPS/commit/859764d2ce4fc3359c4fe4e243a89466fc5aa996
**Flavor:** Fullrelease

**Problem description:**

Layout degradation at AAPS portal overview: unexpected empty space for _BG Check_ events.

**Logfiles**
[AndroidAPS_LOG_1733905150983.log.zip](https://github.com/user-attachments/files/18092689/AndroidAPS_LOG_1733905150983.log.zip)

MilosKozak, 2024-12-11:
screenshot?

olorinmaia, 2024-12-11:
AAPSClient look fine.
[image]

AAPS
[image]

vanelsberg, 2024-12-11:
> screenshot?

[image]

MilosKozak, 2024-12-11:
fixed by https://github.com/nightscout/AndroidAPS/commit/6f3e9372ef73045ec88c1fc389823d8fcc2ef17b

closed 2024-12-11 by MilosKozak