nightscout/AndroidAPS #3608: [DEV] 3.3.0-dev-e doesn't broadcast data to xdrip

Pentagit, 2024-12-11:
I updated to 3.3.0-dev-e today (Build 6f3e9372ef) and noticed that bg-values. smbs and the like are no longer being broadcasted over to xdrip. Haven't changed any settings.

olorinmaia, 2024-12-12:
It works for me. Are you sure all settings are correct?

Pentagit, 2024-12-12:
Yes since I haven't changed any settings, just updated aaps with the newer dev version and broadcast stopped working. I'll try dev-f tomorrow and will get back to you thanks

olorinmaia, 2024-12-12:
I activated xDrip sync  yesterday. I can see basal rendering in xDrip isn't updated, but bg, treatments and profile are synced. Not sure if it takes longer time for basal rendering to work.

MilosKozak, 2024-12-12:
dev-f works