nightscout/AndroidAPS #3613: Dev F Progression barre stuck

robertrub, 2024-12-12:
@MilosKozak As we talked on Discord, with the last dev f, the progression barre gets stuck till next BG reading and stops data being sent to watch.

[image]

[image]

Pulling on the Loop tab gives a black screen on loop and AutoISF

[image]

[image]

The problem seems only with AutoISF (but that needs to be tested) and since the arrival of Dev F. Last dev E was working OK.

Here are the logs. See around 19:11

[image]

vanelsberg, 2024-12-12:
FYI: Dev-e, commit#6f3e937, "fix careportal layout" is ok.

Josch24, 2024-12-12:
Dev-e was ok till 09.12.2024. Dev-e from 10.12.2024, has the Problem. Picture is from last dev-e, that was working ok. 
[image]

MilosKozak, 2024-12-12:
does it creash without AutoISF?

robertrub, 2024-12-12:
Nope. I switched to OpenSMB to test and it seems to work ok. Didn't check with the watch really but no more progression barre stuck on the main UI.

Josch24, 2024-12-12:
Without autoISF, on Standard SMB, everything is ok. Also Watch is getting BG immediately.

vanelsberg, 2024-12-12:
Confirming: Have been running OpenSMB/DASH for one hour now. No issue...

robertrub, 2024-12-12:
@MilosKozak Just had the _black screen then the UI "refilling"_  when BG came in and the loop kicked in **but** I was using OpenSMB !!!

So, not only an AutoISF problem...

MilosKozak, 2024-12-13:
i don't see the issue. Please go commit by commit to find out which broke it for you

closed 2024-12-15 by robertrub