nightscout/AndroidAPS #3617: Bluetooth connection issues AAPS 3.2.04 and Accu-Chek Insight pump and with Dexcom G7 CGM (Samsung A54B with Android 14)

PieterGit, 2024-12-14:
Context:
* Samsung A54B with Android 14 
* AAPS 3.2.0.4 Master 
* CGM Dexcom G7 via Official Dexcom G7 app. Xdrip as Companion app
* Accu-Chek Insight pump

I'm having a lot of(bluetooth) connection issues between both APS-Insight Pump and the Samsung Phone and Dexcom G7 CGM.
I seems to be more severe with Android 14, and especially since the "Samsung 1 November 2024 security upgrade"
(OneUI 6.1, Android version 14,  A546BXXUBCXK1 / A546BOXMBCXK3 / A546BXXUBCXK3).

I will investigate logs further and see if I can get more details.

MilosKozak, 2024-12-16:
provide logfiles

alfureu, 2025-01-09:
Pixel 7 with GrapheneOS build 2025010700 (updated today), I lost connection to Insight with AAPS 3.3.1.0. Some logs below, if it helps...

```
01-09 09:55:50.856 16312 16312 D CORE    : [main]: [MainApp.onCreate():92]: onCreate
01-09 09:55:50.861 16312 16312 I Choreographer: Skipped 99 frames!  The application may be doing too much work on its main thread.
01-09 09:55:50.873 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [MainApp$onCreate$1.invokeSuspend():108]: Version: 3.3.1.0
01-09 09:55:50.875 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [MainApp$onCreate$1.invokeSuspend():109]: BuildVersion: 2ac634c908-2025.01.07
01-09 09:55:50.876 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [MainApp$onCreate$1.invokeSuspend():110]: Remote: https://github.com/nightscout/AndroidAPS
01-09 09:55:50.881 16312 16312 D CORE    : [main]: [DummyService.onCreate():42]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0 flags=ONGOING_EVENT|ONLY_ALERT_ONCE color=0x00000000 category=status vis=PRIVATE)
01-09 09:55:50.883 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadSettings():109]: Loading stored settings
01-09 09:55:50.885 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [PluginBase.setPluginEnabled():81]: Starting: Ongoing Notification
01-09 09:55:50.890 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_GENERAL_PersistentNotificationPlugin_Enabled:true
01-09 09:55:50.891 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_GENERAL_PersistentNotificationPlugin_Visible:false
01-09 09:55:50.893 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [PluginBase.setPluginEnabled():81]: Starting: Overview
01-09 09:55:50.900 16312 16312 D CORE    : [main]: [DummyService.onStartCommand():68]: Starting DummyService with ID 4711 notification Notification(channel=AndroidAPS-Ongoing shortcut=null contentView=null vibrate=null sound=null defaults=0 flags=ONGOING_EVENT|ONLY_ALERT_ONCE color=0x00000000 category=status vis=PRIVATE)
01-09 09:55:50.902 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_GENERAL_OverviewPlugin_Enabled:true
01-09 09:55:50.903 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_GENERAL_OverviewPlugin_Visible:true
01-09 09:55:50.905 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [PluginBase.setPluginEnabled():81]: Starting: IobCobCalculator
01-09 09:55:50.909 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_GENERAL_IobCobCalculatorPlugin_Enabled:true
01-09 09:55:50.910 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_GENERAL_IobCobCalculatorPlugin_Visible:false
01-09 09:55:50.913 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [PluginBase.setPluginEnabled():81]: Starting: Actions
01-09 09:55:50.914 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_GENERAL_ActionsPlugin_Enabled:true
01-09 09:55:50.915 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_GENERAL_ActionsPlugin_Visible:true
01-09 09:55:50.917 16312 16328 D CORE    : [DefaultDispatcher-worker-1]: [PluginBase.setPluginEnabled():81]: Starting: Rapid-Acting Oref
01-09 09:55:50.918 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_INSULIN_InsulinOrefRapidActingPlugin_Enabled:true
01-09 09:55:50.919 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_INSULIN_InsulinOrefRapidActingPlugin_Visible:false
01-09 09:55:50.920 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_INSULIN_InsulinOrefUltraRapidActingPlugin_Enabled:false
01-09 09:55:50.922 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_INSULIN_InsulinOrefUltraRapidActingPlugin_Visible:false
01-09 09:55:50.923 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_INSULIN_InsulinLyumjevPlugin_Enabled:false
01-09 09:55:50.924 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_INSULIN_InsulinLyumjevPlugin_Visible:false
01-09 09:55:50.925 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_INSULIN_InsulinOrefFreePeakPlugin_Enabled:false
01-09 09:55:50.926 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():129]: Loaded: ConfigBuilder_INSULIN_InsulinOrefFreePeakPlugin_Visible:false
01-09 09:55:50.927 16312 16328 D CONFIGBUILDER: [DefaultDispatcher-worker-1]: [ConfigBuilderPlugin.loadPref():123]: Loaded: ConfigBuilder_SENSITIVITY_SensitivityAAPSPlugin_Enabled:false
[... cut ...]
01-09 09:56:24.445 16877 16885 I cout.androidaps: Background concurrent mark compact GC freed 48MB AllocSpace bytes, 19(828KB) LOS objects, 75% free, 14MB/58MB, paused 342us,1.483ms total 118.124ms
01-09 09:56:26.672 16877 16981 D ProfileInstaller: Installing profile for info.nightscout.androidaps
01-09 09:56:27.197 16877 16972 D PUMPQUEUE: [InsightPluginHandler]: [CommandQueueImplementation.add():191]: Adding: CommandReadStatus - READSTATUS Pump driver changed.
01-09 09:56:27.224 16877 16913 D PUMPQUEUE: [CommandQueueImplementationHandler]: [CommandQueueImplementation.notifyAboutNewCommand$lambda$6():239]: Work is already running
```

Philoul, 2025-03-09:
@PieterGit Do you still have a lot of connection issue with your Insight pump?
I'm not sure if I correctly understood your comment talking about Insight pump and Dexcom G7 🤔
If you lost connection with both devices (pump and CGM sensor), you should probably look within android config...
I don't see anything relevant with copy of logfile content (log should be copied immediatly and PUMP should be enabled in Maintenance > Log Settings to record information).
If now issue is fixed (with Android update /tuning ...) can you please close this issue, or give more information?

PieterGit, 2025-04-09:
Sorry, for not responding. The problems were with the APS-pump (Insight) connection. 
The issue has been resolved with one of the optimazations of the phone such as 
Good Guardians or Galaxy app booster. See more tips at https://dontkillmyapp.com/samsung
I cannot reproduce it anymore and I am affraid the PUMP log setting was not enabled :(
Closing this issue because it's not reproducable anymore and no log exists.
If others have solver problems with APS-Pump communication and have more info, please reopen or add a new issue and perhaps link this issue.

closed 2025-04-09 by PieterGit