nightscout/AndroidAPS #3620: [Dec] NSv3 is spamming/duplicating entries in log

olorinmaia, 2024-12-14:
AAPS 3.3.0-dev-f
Build: b9686af643-2024.12.14
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 15.0.3
Engineering mode enabled 

Logs
AAPS 
[AndroidAPS_LOG_1734180646075.log.zip](https://github.com/user-attachments/files/18135892/AndroidAPS_LOG_1734180646075.log.zip)

AAPSClient 
[AndroidAPS_LOG_1734180660737.log.zip](https://github.com/user-attachments/files/18135898/AndroidAPS_LOG_1734180660737.log.zip)

I've seen this happen with TBR and now TT. Not sure what causes it to happen but attaching logs. 
AAPSClient and AAPS is running on same phone.

It makes the sync slower between client and AAPS. 

[image]
[image]
[image]

olorinmaia, 2024-12-14:
Also tried changing to NSV1 on all phones and a loop also happens there. Eventually it will stop but it takes a long time.

[AndroidAPS_LOG_1734213947732.log.zip](https://github.com/user-attachments/files/18137922/AndroidAPS_LOG_1734213947732.log.zip)

https://github.com/user-attachments/assets/bd4e923f-f2f3-429d-8aca-d5bb3b7c9eb4

MilosKozak, 2024-12-17:
did you start TT in one app and cancel in another ?

olorinmaia, 2024-12-17:
Perhaps, I had an automation that set 5,5 mmol/l for 24 hours. I don't know if it is related but i'm not able to reproduce this on the new NS server.

MilosKozak, 2024-12-26:
https://github.com/nightscout/AndroidAPS/commit/6a670611df1dc8a87c9b6300c1a632fe6dc358d6

emilisev, 2024-12-29:
Still happening on dev-g with this fix.
In attached log, for ex search for `2024-12-29T11:50:18.992Z` and `2024-12-29T11:50:18.924Z`
[AndroidAPS.log](https://github.com/user-attachments/files/18269505/AndroidAPS.log)

olorinmaia, 2024-12-29:
Isn't this related to https://github.com/nightscout/AndroidAPS/issues/3571 which isn't closed?

emilisev, 2024-12-29:
Yes, #3571 is still a current bug for me too. I was hoping to see both resolved at the same time.

MilosKozak, 2024-12-29:
from my point of view it's correct. Status is generated on more places (not only after enact) for example for cases like open loop, failed enact etc.
I added logging anyway so you can see the reason https://github.com/nightscout/AndroidAPS/commit/5e706888e0982b8e5a626a3c8a3afe27217523d9

closed 2024-12-29 by MilosKozak

emilisev, 2024-12-30:
This is a regression compared to 3.2, we did not have the issue of duplicate records, and duplicate info showing in AAPSClient. This will clutter people's NS database. For me, this is causing issue to an app I have based on NS API cause it can't download all records for 1 day, because there are too much.

olorinmaia, 2024-12-30:
This is related to https://github.com/nightscout/AndroidAPS/issues/3571

It's true this wasn't an issue in 3.2. I wonder if it's possible to combine the records or filter out in AAPSClient so that only the most recent device status is shown?

It's harder to read as it is now. There is no good reason to show multiple device statuses with same information sent at same time or show device statuses from a long time ago. 
[image]

emilisev, 2024-12-30:
Here's the log with extra debug. Look for `2024-12-30T10:00:44.722Z` and `2024-12-30T10:00:48.991Z`.
Here's the outline : 
[image]
[AndroidAPS.log](https://github.com/user-attachments/files/18273741/AndroidAPS.log)

emilisev, 2025-01-05:
seems solved on latest dev

olorinmaia, 2025-01-05:
I'm using v3 and are still getting duplicate OpenAPS entry in AAPSClient. Are you using v1?

emilisev, 2025-01-05:
Yes I'm using v1. I still have the duplicate OpenAps entry in AAPSClient as well, seems they were 2 duplicate issues after all... I always have one older than the other. One is the regular <5mn old, the other one is generally between 15 and 40mn old... I'll try to investigate the logs tomorrow and see if I can find something useful.