nightscout/AndroidAPS #3622: [Dev] NSv3 full sync isn't getting boluses except for SMB

olorinmaia, 2024-12-14:
AAPSClient2 3.3.0-dev-f
NSCLIENT
Build: b9686af643-2024.12.14
Remote: https://github.com/nightscout/AndroidAPS
Flavor: aapsclient2release
Nightscout version: 15.0.2
Engineering mode enabled 

Logs
[AndroidAPS_LOG_1734183665898.log.zip](https://github.com/user-attachments/files/18136095/AndroidAPS_LOG_1734183665898.log.zip)

First did a fresh install or clear database of AAPSClient. The Doing a full sync on AAPSClient with NSv3. Expect to get same iob as AAPS. Seem to be missing boluses except for smb. 

[image]

AAPS

[image]

olorinmaia, 2024-12-14:
AAPS is running and getting bg but AAPSClient is spamming in the logs and not updating. Attaching more logs.
[AndroidAPS_LOG_1734184080437.log.zip](https://github.com/user-attachments/files/18136119/AndroidAPS_LOG_1734184080437.log.zip)

Video
https://photos.app.goo.gl/7yNNiW2TZxfUnSXq6

MilosKozak, 2024-12-16:
nothing interresting in logs
if it's replicable pm me NS credentials

olorinmaia, 2024-12-16:
Pm'd you credentials, but no. NSV3 log say insert bolus, but not recorded to NS treatments or displayed on graph.

MilosKozak, 2024-12-16:
this is probably NS or mongo issue

MilosKozak, 2024-12-16:
for me it works

olorinmaia, 2024-12-16:
Related to https://github.com/nightscout/AndroidAPS/issues/3623 I think

MilosKozak, 2024-12-17:
i did a few tries (bolus wizard) using virtual pump a everything is synced correctly to Client

closed 2024-12-17 by olorinmaia