nightscout/AndroidAPS #3623: [Dev] NSv1 - All treatments doesn't sync between aaps and client

olorinmaia, 2024-12-15:
AAPS 3.3.0-dev-f
Build: b9686af643-2024.12.14
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 15.0.2
Engineering mode enabled

When entering bolus calc in aaps I've noticed it doesn't sync over to client in all cases. See example below.

Logs.
AAPS
[AndroidAPS_LOG_1734221096370.log.zip](https://github.com/user-attachments/files/18138253/AndroidAPS_LOG_1734221096370.log.zip)

AAPSClient
[AndroidAPS_LOG_1734221112371.log.zip](https://github.com/user-attachments/files/18138254/AndroidAPS_LOG_1734221112371.log.zip)

Screenshot
AAPS
[image]
[image]

AAPSClient
[image]
[image]

MilosKozak, 2024-12-16:
I keep it open but next step will be remove v1

olorinmaia, 2024-12-16:
Do you plan on that for 3.3? I understand there is alot more complexity providing support / code compability for two different sync api for NS.

Main problem many have with v3 is the basal rendering an inability to use NS to enter treatments. F.ex. parents got Iphones and can't use AAPSClient. All of this is known but it's not clear when it will be fixed or supported within NS.

This bug seem to be 100% reproduceable and does not occur in 3.2, so if NSv1 ships with 3.3 this will cause alot of wrong iob between main/client.

MilosKozak, 2024-12-16:
3.3 will have v1

MilosKozak, 2024-12-17:
i did a few tries (bolus wizard) using virtual pump a everything is synced correctly to Client using v3

olorinmaia, 2024-12-17:
I've noticed small boluses get through more often than big. So test both small 0,1 bolus Calc and bigger >0,5.

I think this can be closed since it sync between v3 regardless of uploading to ns. Virtual pump bolus calc isn't used much by us anyway. We bolus through AAPS dash driver and wear app which works 100% of the time.

This could lead to wrong iob in some edge cases with v1 but when it's so hard to reproduce we should probably just close it. And hope we can remove v1 for next release.

closed 2024-12-17 by MilosKozak

rorybladen, 2025-01-13:
I think we may have just had this occur with v3. BG was syncing but a carb/bolus entered with the bolus calculator did not sync. The treatments were not in nightscout so Aaps client can likely be excluded from the issue.

The missed treatment sync was an issue because it wasn't spotted due to no break in BG value sync which suggested nightscout/mongo was operational and wasn't a cause for concern. Following that and before a full sync was attempted, the iob/cob in aaps clients was incorrect due to the missed treatment data.

A full synchronization restored nightscout and Aaps client to match Aaps iob/cob.

During troubleshooting 10 carbs were entered into an aaps client and synced across to the master aaps. The carbs were then removed on the master aaps using the treatments history screen. The removal synced back to the aaps clients correctly. However.. the 10 carb treatment was still visible in Nightscout. The carbs didn't count towards the iob displayed in nightscout so it appears as just an artifact - guessing deleted treatments are just tagged as deleted but remain in the database, so perhaps a bug with nightscout displaying deleted treatments even though they have no remaining effect on calcs.
Full synchronization also failed to remove the 10 carbs from the nightscout display. The 10 carb treatment artifact was removed using the nightscout edit - drag - delete UI.

Aaps 3.2.0.4
Nightscout 15.0.2

MilosKozak, 2025-01-13:
Nightscout doesn't support v3 protocol internaly. That's the reason why you cannot add entries though NS UI, it doesn't handle remove entries etc (while AAPS is using v3 protocol)

rorybladen, 2025-02-01:
Thanks for the info regarding the v3 protocol.
Regarding the original issue of the Bolus which did not upload to nightscout we will keep an eye out for it occurring again.
The concern is that AAPSClient showed an incorrect IOB/COB.  If AAPS master goes offline it will resync UP once back online. 
So whilst it is possible there was an interruption during the sync directly following the treatment, something then also prevented this particular treatment resyncing on the subsequent uploads. 
The missing treatment in AAPS is explained because the treatment wasn't present in Nightscout, however it doesn't explain the IOB/COB mismatch.
Does AAPSClient sync actual IOB/COB values or does it just sync treatments/BG and run its own loop in parallel? (this appears to be what happens?)  If so would it make sense to also sync the IOB/COB and check it against the AAPSClient loop derived values to highlight any discrepancy?
Thanks.