nightscout/AndroidAPS #3624: [3.3.x. Beta] Client 2, COB displayd negaitve after neg. Carb Correction, Broadcasting to Client 1 for Dual Wear display won't work after that

gerison77, 2024-12-15:
AAPSClient2 3.3.0-beta1
NSCLIENT
Build: 9209de1dd0-2024.12.11
Remote: https://github.com/nightscout/AndroidAPS
Flavor: aapsclient2release
Nightscout-Version: 15.3. dev

Rig Master: N6000 - Diakem G7 - G7, Android 12, Ticwatch Pro 3, Wear 3.5

Some Time after entering neg. carbs. COB was displayed negative on the graph. Some time later Client 2 stopped broadcasting to Client 1, no data of Client 2 in dual watch face. 

I did a full database reset, which didn't work. After a reinstall of Client 2, broadcasting was wrong again.

[image]

Logs are sent via email

Philoul, 2024-12-15:
Log should be sent to your own mail, then files attached here.
You can also connect phone on computer and copy log files.
When you have this kind of issue o. AAPSClient, having the equivalent screenshot on main app of your child can help analysis...

gerison77, 2024-12-16:
Logs added. Would be a screenshot from history of mein app suffice?

MilosKozak, 2024-12-16:
this is fixed in dev and will be part of next beta

closed 2024-12-16 by MilosKozak