nightscout/AndroidAPS #3639: AAPS Client/ Websockets / BZ Sync

elgrupo123, 2024-12-20:
Seit der Beta 1 Version, auch die Devs Versionen,  werden bei aktiven Websockets irgendwannmal die BZ Werte nicht mehr an AAPS Client übertragen. Der Rest geht durch. 
Master-Websockets EIN, AAPSCLient1-Websockets AUS-funktioniert einwandfrei, aber langsam. AAPSCLient2 Websockets AUS-funktioniert nicht einwandfrei, bricht irgendwannmal ab.
Seit ca. 6.20 Uhr gabs kein Sync der BZ Werte mehr. NS vollständig synchronisieren behebt den Fehler

[AndroidAPS.log](https://github.com/user-attachments/files/18207147/AndroidAPS.log)
[image]

elgrupo123, 2024-12-20:
> Which ns server / hosting are you using? Version of NS?
> 
> I don't experience this with 10be.de NS master or dev.

Hi,
10be.de + NS Master 15.0.2.
Bei mir mit aktivierten WS kommt der Fehler alle paar Sunden vor, auf dem Follower meiner Frau läuft es mit aktivierten WS garnicht.

elgrupo123, 2024-12-20:
Und hier die Log Dateien von unserem zweiten Follower, bei dem es mit den aktivierten Websockets überhaupt nicht synchronisiert wird.
[AndroidAPS.log](https://github.com/user-attachments/files/18213885/AndroidAPS.log)

Philoul, 2024-12-20:
Write in english please.

olorinmaia, 2024-12-20:
Ok, so if I want to try to reproduce this, to confirm, is your setup like this?
AAPS = NSClientV3 + WS = ON
AAPSClient 1 and 2 = NSClientV3 + WS = OFF ? 

And just one of them sync? For me it sounds like one of your phones got some issues with either settings or permissions. But I will try this setup myself and see.

My regular setup is WS = ON with NSclientv3 on all phones.

elgrupo123, 2024-12-20:
Not quite right.

All with NS V3

Master with NS V3 - WS Active - no problems.

Follower 1 - with AAPS Client (WS Active) problems after a few hours, AND with AAPS Client2 (WS deactivated), no problems.

Follower 2 - with AAPS Client (WS Active), problems straight away. Only works with WS deactivated.

gerison77, 2024-12-21:
Running 2 Rigs with the follwoing setups

Each Rig is running: Master Blackview N6000 3.3.0-rc1 NSClient V3 websockets on and Client Blackview BV 9200 3.3.0-rc1 NSClient V3 websockets on. Servers hostet on 10be.de running **Nightscout 15.3 dev**

Can't confirm that issue in this setup on 2 rigs.

elgrupo123, 2024-12-21:
That doesn't help me. I can only report my problems. 
Just switched all devices to 3.2.0.4 and the problem no longer exists. 
I posted the logs, I can't contribute anything more. I'm not a developer.

olorinmaia, 2024-12-21:
Strange it works when you downgrade to 3.2 🤔

I have no issues with two phones for my testrig with 3.3 dev. For the sake of it I'll test with three phones later today. If I encounter the same I'll also share logs.

Philoul, 2024-12-21:
Can you confirm this issue can be reproduced 🤔.
I mean (was not a temporary issue that can arrive from time to time in NS server, and fixed a bit later...)
Now it works again (with 3.2.0.4), if you update rig to latest RC1, it reproduce the issue or not?
@MilosKozak any other idea of potential root cause?

elgrupo123, 2024-12-21:
> Can you confirm this issue can be reproduced 🤔. I mean (was not a temporary issue that can arrive from time to time in NS server, and fixed a bit later...) Now it works again (with 3.2.0.4), if you [...]

from 3.2.0.4 back to 3.3.0 does not help. The only way to get the value is to synchronize NSV3 manually, then a value comes, after that nothing more.

olorinmaia, 2024-12-21:
Can you switch to NSClientv1 on all phones to see if problem persist?

Also can you take a screenshot of your apiv3 settings in 10be server settings? At the very bottom

elgrupo123, 2024-12-21:
> Can you switch to NSClientv1 on all phones to see if problem persist?
> 
> Also can you take a screenshot of your apiv3 settings in 10be server settings? At the very bottom

[image]

olorinmaia, 2024-12-21:
@elgrupo123 can you try to switch to ns master branch in 10be?

elgrupo123, 2024-12-21:
> @elgrupo123 can you try to switch to ns master branch in 10be?

It is set to NS Master the whole time. A test with Dev didn't help.

olorinmaia, 2024-12-21:
It stopped syncing bg on one of the phones with dev branch on 10be. Then I switched to master and since then it has been working. Could you try clear database in the AAPSClients and do a full sync after when using master branch of NS?

I'm suspecting something from NS dev is causing this, but please let me know if that helps.

olorinmaia, 2024-12-21:
Confirming this with 10be master branch and 3 phones. V3 and Ws = on
Log AAPSclient
[AndroidAPS_LOG_1734793839136.log.zip](https://github.com/user-attachments/files/18218906/AndroidAPS_LOG_1734793839136.log.zip)

AAPS
[AndroidAPS_LOG_1734793804048.log.zip](https://github.com/user-attachments/files/18218908/AndroidAPS_LOG_1734793804048.log.zip)

Screenshot of AAPSClient missing bg updates. Other data is syncing. Other phones are ok. I can see in v3 log that some treatments are recurring / spamming with older time than now. 

[image]
[image]

elgrupo123, 2024-12-21:
I have activated another follower (3 in total) and have the same problem. 
With V1 they all work correctly and I'll leave it like that for now.

olorinmaia, 2024-12-21:
All my phones are older android than 15. I dunno.

gerison77, 2024-12-21:
Running now 3 clients as well in NSClient V3. No Sync Problems with BG since 1 hour. But age of pump is not synced (3rd phone was not activated since two days and missed the pump change, maybe that is the reason). All Phones run Andorid 12 and 3.3.0-rc1

Philoul, 2024-12-21:
> from 3.2.0.4 back to 3.3.0 does not help. The only way to get the value is to synchronize NSV3 manually, then a value comes, after that nothing more.

I got exactly this issue during tests a few weeks ago, root cause was AAPS phone set with NSClient V1, and AAPSClient phone was with NSClient V3... 🤔

elgrupo123, 2024-12-21:
Apparently it's just the websockets. If I deactivate the websockets on all devices, it works without any problems even with V3. 
I can live with that, the issue can be closed.

olorinmaia, 2024-12-21:
Well. Without websockets updates are slower. I wouldnt say this is a solution to the problem.

olorinmaia, 2024-12-22:
NSClientv1 > NSClientv3

elgrupo123, 2024-12-22:
> NSClientv1 > NSClientv3

?

anko184, 2024-12-24:
I'm having the exact same problem as soon as I connect a third phone. With two phones as followers everything worked without any problem.
I'm running 3.3.0-rc1 on master and aapsclient2 on the followers. Master phone is running Android 14, 1 follower the same and the others are on Android 15 QPR2b1. The problem started when I connected the second Android 15 phone. 
I'm going to remove one A15 phone to check if the problem is still there and I'll report here.

elgrupo123, 2024-12-25:
> I'm having the exact same problem as soon as I connect a third phone. With two phones as followers everything worked without any problem. I'm running 3.3.0-rc1 on master and aapsclient2 on the [...]

My 1 follower phone runs on Android 15 and has no problems. The 2nd and 3rd followers have Android 14 and don't work.

anko184, 2024-12-25:
> My 1 follower phone runs on Android 15 and has no problems. The 2nd and 3rd followers have Android 14 and don't work.

I think it has to do with AAPS itself and not the phones. After 8-10 hours since I removed one A15 phone everything works as expected with rc1 and WS on.

MilosKozak, 2024-12-26:
I found a bug in @olorinmaia case. NSCV1 was not properly shutdown after switching v1 -> v3. This is fixed by https://github.com/nightscout/AndroidAPS/commit/6a670611df1dc8a87c9b6300c1a632fe6dc358d6

I run AAPS + two clients with WS on and found no issue so far
Update to latest dev, turn on WS and if this happen again create new issue with description, time and logs from all 3 devices

closed 2024-12-26 by MilosKozak

MilosKozak, 2024-12-26:
another possible issue is wrong/different date on all phones

elgrupo123, 2024-12-29:
The latest Dev version (67caa1b) no longer synchronizes with AAPS Client. 
Here are the log files.
[Follower_AndroidAPS.log](https://github.com/user-attachments/files/18269069/Follower_AndroidAPS.log)
[MASTER_AndroidAPS.log](https://github.com/user-attachments/files/18269070/MASTER_AndroidAPS.log)