nightscout/AndroidAPS #3642: Double arrows : logs

robertrub, 2024-12-21:
@MilosKozak You had asked the logs when double arrows arrive in AAPS.

I think they happen when there is a steep direction change in BGs. This nights examples : sensor compression, goes low, I turn or wake up with the alerts and BGs raise. I have the last dev and am using average smoothing. G6 and xDrip.

Arrows are OK in xDrip

[image]

[image]

[image]

The logs during the double arrows (look around 03:17 

[image]

And a bit after the double arrows

[image]

robertrub, 2024-12-21:
Noisy sensor. Happened again

[image]

[image]

[image]

[image]

The logs

[image]

robertrub, 2024-12-22:
This shity sensor is very sensible to compression and wakes me up every night :( But I hope shows the reason to the double down arrows in the multiple logs ;)

[image]

[image]

[image]

robertrub, 2024-12-23:
Another case but with sensor errors. The sensor stopped sending BG for 30 min then started sending. Arrow normal on xDrip but double down in AAPS

[image]

[image]

[image]

Came back to normal after 6 BG 

[image]

Logs after the first doublé arrow

[image]

MilosKozak, 2024-12-25:
03:17:14.302 [DefaultDispatcher-worker-3] D/BGSOURCE: [XdripSourcePlugin$XdripSourceWorker.doWorkAndLog():105]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=4.3345181016144415E-5, com.eveningoutpost.dexdrip.Extras.VersionInfo=2124101120 9f769df-2024.10.11, com.eveningoutpost.dexdrip.Extras.Raw=0.0, com.eveningoutpost.dexdrip.Extras.Display.Units=mg/dl, com.eveningoutpost.dexdrip.Extras.Time=1734747431118, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=-1, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=null, com.eveningoutpost.dexdrip.Extras.SensorStartedAt=1733386202079, com.eveningoutpost.dexdrip.Extras.Noise=-9999.0, com.eveningoutpost.dexdrip.Extras.SensorBattery=-1, com.eveningoutpost.dexdrip.Extras.BgSlopeName=**DoubleDown**, com.eveningoutpost.dexdrip.Extras.BgEstimate=97.0, com.eveningoutpost.dexdrip.Extras.SourceDesc=G6 Native / G5 Native, com.eveningoutpost.dexdrip.Extras.SourceInfo=G6 Native, com.eveningoutpost.dexdrip.Extras.Collector.NanoStatus=  }]

xdrip is sending DoubleDown. AAPS doesn't change the arrow if provided.
Report to xDrip

robertrub, 2024-12-25:
Ok, tnx. It's funny how then send double down arrows but show only one up...

Please leave this issue open. I'll link it to them.

robertrub, 2024-12-26:
@MilosKozak This is the answer from Navid / xDrip. They say that xDrip transmits the arrows without treatment but calulate the arrow on their own UI.

Can AAPS do the same ? We already have the delta, needs some code to determine the angle 😉

https://github.com/NightscoutFoundation/xDrip/issues/414#issuecomment-2155740131

MilosKozak, 2024-12-26:
i don't think we should have different behavior for diffrent BG sources.
xDrip should decide what should be presented to AAPS

robertrub, 2025-01-22:
Navid from xDrip isn't interrested in changing anything and thinks that the double arrows come from the transmitter itself. I don't use BYODA and I can’t check if it's the case.

closed 2025-02-21 by robertrub