nightscout/AndroidAPS #3653: When temporarily disconnected with Dexcom G6, backfill data comes well to xDrip+ but not to AAPS when reconnected.

dar-kod, 2024-12-23:
- The precise time the problem occurred: disconnection was between 18:55 to 19:10.

- The circumstances and steps that caused the problem:

_It is important to say that I am a rare user with this problem. The settings in xDrip+ are exactly as described in [Inter-app Settings (Broadcast)] (https://androidaps.readthedocs.io/en/latest/CompatibleCgms/xDrip.html#inter-app-settings-broadcast) with "Identify receiver"._

If you come close to your phone after about 15 minutes, you will see that the data in AAPS is not backfilled when you reconnect.

- The Build version: see attachment. But the version is not important, as this problem has been around for a few years now.

- The app's log files are attached.
[image]
[image]
[logs.zip](https://github.com/user-attachments/files/18232622/logs.zip)

MilosKozak, 2024-12-25:
report to xdrip

olorinmaia, 2024-12-26:
https://github.com/NightscoutFoundation/xDrip/discussions/3831#discussioncomment-11665537

I don't think I can reproduce this issue. Just tested it. Either way AAPS internally back fills in sensor errors etc. So why is it a problem if xdrip doesn't send all in between?

Are you using xdrip as collector or in companion mode?

dar-kod, 2024-12-26:
> So why is it a problem if xdrip doesn't send all in between?

You can try asking yourself what it would be like for you if you did not get backfill data. If others have it, why should I not? It looks better, the statistics are much more accurate and does it affect the loop (not sure).

> Are you using xdrip as collector or in companion mode?

As a collector.

Philoul, 2024-12-26:
It works fine with BYODA...

If you use xdrip instead of BYODA, you should accept xdrip limitations, or open the issue within xdrip project 🧐, or open the issue within xdrip project.
AAPS will not replace received data from xdrip by something else...

I don't understand what's your expectation with this issue opened in AAPS project 🤔.

MilosKozak, 2024-12-26:
BOYDA always send last XX values (as they may change in the past because of smoothing) and AAPS accepts all changes
xDrip is probably sending always only last value

dar-kod, 2024-12-26:
> I don't understand what's your expectation with this issue opened in AAPS project 🤔.

I was not sure that this problem needed to be solved by xDrip. When Milos informed me, I opened the topic in the [xDrip discussions](https://github.com/NightscoutFoundation/xDrip/discussions/3831).

closed 2024-12-26 by dar-kod