nightscout/AndroidAPS #3655: [DEV] Medtrum settings does not show current setting

jbr7rr, 2024-12-25:
dev 0a29d84c4f96e04e39391077239af9d3c52c1a7

In the settings screen the current setting is not shown anymore:
[image]

Compare this to the screenshot in the wiki (which is master 3.2):
[image]

I have looked into the code, but could not find any obvious reasons why this is happening

MilosKozak, 2024-12-25:
migration from xml to `BooleanKey`... style will resolve it

closed 2024-12-25 by MilosKozak

jbr7rr, 2024-12-29:
This is not fixed, PR was for a different issue

robertrub, 2024-12-29:
I confirm, on last dev, the information is only visible when I tab on the lines and nothing gappens when I tap the serial number line

[image]