nightscout/AndroidAPS #3659: 3.3 rc1 stop insulin dosing below bg value doesn't work with "tdd based sens ratio for basal and target" enabled

RalfCGM, 2024-12-28:
See title. 

Basal kept running, while bg was below threshold.
Turning of "  tdd based sens ratio... “immediately stopt basal as expected based on bg level.

MilosKozak, 2024-12-29:
this is not enough information to be able check anything, provide screenshots, logs, time  etc

RalfCGM, 2024-12-29:
`[
[AndroidAPS_LOG_1735486309286.log.zip](https://github.com/user-attachments/files/18270098/AndroidAPS_LOG_1735486309286.log.zip)
[image]
](url)`

olorinmaia, 2024-12-29:
@RalfCGM Screenshot or log isn't viewable.

MilosKozak, 2024-12-29:
history is not enough, take a screenshot of overview when this happen

tim2000s, 2024-12-30:
It's unlikely that this has overridden any threshold settings, given what it does. It's more likely that the modifications that the button makes to ISF calculation made the oref1 algorithm calculate values higher than target, with enough other factors valid, that it dosed insulin. 

You'd need to provide a screenshot of the decision making to be able to understand it any further.

RalfCGM, 2025-01-01:
Since it appeared a little dangerous, and I'm not having a lot of energy left over I descided to leave the setting off for know. Resulting in understandable behavior on low bg values.

Ill update to stable release soon and try to recreate te issue if I have time and energy for it. Ill keep you posted.

RalfCGM, 2025-01-01:
> @RalfCGM Can you Show IOB on the history graph in same time period. If IOB is negative Low glucose suspend can give insulin.

Attached. Abs insulin was positive. [
[image]
](url)

olorinmaia, 2025-01-01:
It's IOB graph you need to show. You can have negative IOB allthough ABS is positive

RalfCGM, 2025-01-02:
Sorry, then Dutch translations must be a little if. The graph you need is shown as *"opgenomen insuline" which translates as "included insulin". Didnt understand that this was iob. But indeed this graph shows a really minor amount of negative iob. And as I understood from you, that could be expected bahavior?

olorinmaia, 2025-01-02:
Yes that is the way low glucose suspend work as far as I understand it. 

But once iob is positive insulin should be suspended below threshold. If you experience this again try catching screenshot of overview screen and add logs. 👍

In crowdin you can apply to join the German translation team and try to find this string so it is more precise 😉

closed 2025-01-03 by MilosKozak