nightscout/AndroidAPS #3662: RC 3.3.0 Bugs that I found,

Alamo04, 2024-12-29:
Using AAPS DynISF ON  Pixel 5, was on RC 3.3.0 since it goes online and Come from Master.

I'm realy upset about RC 3.3.0, in all cases worse compared to Master.

Alarm after one Missing CGM value, not Sure that is a Bug or intend.
The sound comes from AAPS, with a associated message.
I have the complete message as a screenshot and I had to be really fast for that, because the next value comes after seconds and cleared this Message.

[image]

Screenshot_20241229-125126~21048×340 25.4 KB

A new bug has also appeared since yesterday.

The warning that the canula has been Not changed for too long, the color turns red. This remains after the canula is changed. Restarting and simulating another canula change doesn't Help. This Bug comes yesterday after 7-8 days using RC.

[image]

Screenshot_20241229-1028011080×276 20.1 KB

But these are small bugs, we don't need to talk about them any further.

The more important thing is that the algorithms no longer work for me.

 - TIR of around 88%, up from around 95% previously

- TDD/7D of 49, up from 42 previously

- far more inconsistent blood sugar trend

- the new "optional"  sensitivity comes out of hell. It leads to horrible algorithmen settings. Deactivated after one day using. Should that sensitivity not be used with DynISF ?

I'm now back to Master, and must again wait 7 days for enough data for DynISF, because the Database IS again inkompatibel.

For me, it's all a step backwards compared to the master.
And I specifically carried out a new installation, not an update, so that the RC would run smoothly.

Log will be send in a few minutes, because I'm back to Master. I hope I send the one with RC running (hopefully).

olorinmaia, 2024-12-31:
@MilosKozak 
I agree that the notification about fallback to profile sensitivity because of 1 missing bg is abit to much and will annoy most users especially because of the nature of all alerts and stuff that comes with diabetes from before. Sensor error will cause missed readings and I think we should try to find a way not to notify about this until many more bgs are missed. If a few are missed we should still be able to calculate a 24h average ISF.

I tested upgrading to 3.3 then downgrade to 3.2 and you are right database get issues with BG. I solved this by doing a full sync with NSClienv3. I think it's unlikely that people will do this but there is a workaround to the problem if data is backed up at nightscout.

For colors regarding canula you need to check settings in overview and status lights. It could be reset or copied wrong from ns.

blaqone, 2024-12-31:
@MilosKozak  I second everything said above. This is ridiculous.

Alamo04, 2025-01-01:
@olorinmaia 
My AAPS is not allowed to load anything from NightScout.

As I say . You can deleted the last canula change, you can tell AAPS you make a new one with the date and time in the past, or you can do a normal new one. The color remains red. But this happened for me first with a canula change after 7-8 days of using RC, short before the new Master came out. You can fix it only with a clean new installation. Settings are OK, as they are the 7-8 days ago on RC.

But this is only an minor optical thing, mutch more importand are my Issues with the new algorithmen.

Sorry for my horrible english.

Alamo04, 2025-01-01:
Installed Master 3.3
Same Bug with Canula age, as in RC

[image]

olorinmaia, 2025-01-01:
Go to settings > overview > status lights. I guess some of them are set to 0. Can you confirm this? I guess it wasn't like that in 3.2

Alamo04, 2025-01-01:
Thats it, was Set to 0.
But how did that happen, must I Set it back now with every canula change ?
Not realy, because IT happened today and Not Yesterday with the change. 🤔

olorinmaia, 2025-01-01:
Set it manually. If it happens again after next cannula change provide screenshots and logs.

Alamo04, 2025-01-02:
`Set it manually. If it happens again after next cannula change provide screenshots and logs.`

That was already done in my first post.

olorinmaia, 2025-01-02:
I think it happens when upgrading from 3.2 to 3.3, same happend here. But after I stayed on 3.3 and adjusted number back to original setting and did a new cannula change the status light number wasn't reset to 0 on my test-rig at least.

Alamo04, 2025-01-02:
@olorinmaia 
I always did a clean, fresh installation of a new Master, or RC.
Never did updates in that case.
AAPS must change this setting and it does it after a different time it is running.
First I see that Issue was after 7+ days using RC1, now after 1+ day using Master. 

For me, this Issue didn't come up immediately after installing and starting your new AAPS.

closed 2025-01-03 by MilosKozak