nightscout/AndroidAPS #3665: AAPS 3.3.0: Main Menu Button CGM no longer visible/nor activates DiaKEM DEX G7 App (BYODA)

Arne1346, 2024-12-30:
Hi,

on my phone I run just the DiaKEM Dexcom G7 for gathering CGM readings. AAPS is getting those directly from there via the BYODA config.
Till 3.2.0.4 I could activate and display the CGM button on the AAPS main screen. When pressing it brought me straight away to the DiaKEM Dexcom G7 App.

With AAPS 3.3.0 activating the button in AAAPS' config is still possible, but it does not show up on the main screen.

This is really unhandy as I now need to switch to the Dexcom app manually everytime I need to confirm or temporarily disable alarms.
Please make sure that this gets fixed and the CGM button works as previoulsly.

thank you

olorinmaia, 2024-12-31:
Strings where adjusted based on @MilosKozak statement in Discord.
See  https://github.com/nightscout/AndroidAPS/pull/3627

MilosKozak, 2024-12-31:
Impossible for Android 11

closed 2024-12-31 by MilosKozak

Arne1346, 2025-01-15:
Thanks for having realized this "Impossible for Android 11" and already closed issue with AAPS 3.3.1.1. -- works perfectly with Android 15