nightscout/AndroidAPS #3666: New Insight driver fails to deliver boluses

tomdvorak, 2024-12-31:
Boluses are recorded in AAPS as delivered, but the pump (AC Insight) bolus history is empty, see the screenshots below.

More details: AAPS 3.3.0.0 was installed on kid's phone shortly after noon (30.12.) and everything seemed to be working fine. 
But later in the afternoon kid's GL started raising, so in the evening we finally decided to issue a bolus correction manually (at 19:28). The animation in AAPS app was suspiciously fast (zero connection time, no sounds from the pump), so we checked the pump bolus history: the last bolus was delivered at 11:54, i.e. before we switched to 3.3.0.0. We tried again with the same result:  bolus delivery recorded by AAPS ( 0,50U at 19:33, see below), but no bolus delivered by the pump. 

I reinstalled the old version (3.2.0.4) and everything went back to normal. The next two items in the pump history (at 19.51 and 19.58, see below) are boluses delivered by the old driver.

[image]

MilosKozak, 2024-12-31:
I see logged crashes on Kingkong Mini 3 related to memory and android system (not aaps fault)
Identification is not provided so I cannot tell it's you

Philoul, 2025-01-01:
Can you please provide screenshot of Insight plugin ? (of course with 3.3.0.0 version)

tomdvorak, 2025-01-03:
So I reinstalled AAPS 3.3.0.0  and this time it seems to be OK. Sorry for the false alarm.

More details: I reinstalled AAPS 3.3.0.0  and tried again to issue a manual bolus. This time AAPS tried to connect and failed, but the bolus was (correctly) not recorded in AAPS treatment. Retry & phone reboot did not help, but restarting bluetooth  did, so the bolus was finally delivered and recorded in AAPS. I  compared the pump bolus history & AAPS treatments and there is one microbolus at 11:34 recorded in AAPS which is missing in the pump history. This, however, coincides with the time when I reinstalled the app, so it does not seem to be critical. I'll be comparing AAPS & pump records for a few more days, but now it seems to be working fine. So once again, sorry for the false alarm. 
   
[image]

closed 2025-01-04 by tomdvorak