nightscout/AndroidAPS #3668: DynIsf Factor reset to 100%

Koeppchen, 2025-01-01:
After update to version 3.3.0.0 from 3.2.0.4 the DynISF factor is  reset to 100%.
The same happens when I restore a profile created with the Old AAPS Version.

To me this looks like there is missing some migration code to transfer this setting from DynISF plugin to the OpenAPS SMB plugin.

MilosKozak, 2025-01-01:
this setting didn't change between 3.2.0.4 and 3.3.0.0

MilosKozak, 2025-01-01:
what was your previous setting?

Koeppchen, 2025-01-01:
It was set to 99. So it didn't cause any trouble in my case.
But i have also heard from others that it was changed to 100 in their case. And also it was also reset to 100 when i imported an old profile. I can send the profile export if you want.
In my case its no problem, I have just opened the case because i don't want anyone to run into problems.

MilosKozak, 2025-01-01:
you can pm the export + password

closed 2025-01-01 by MilosKozak

MilosKozak, 2025-01-02:
fixed by https://github.com/nightscout/AndroidAPS/commit/50d263264fb7874b804590e2812f769a8bfa2d35
thanks!