nightscout/AndroidAPS #3672: Accu-Chek Combo Page not showing Values with 3.3.0.0

Koeppchen, 2025-01-01:
After Update to Version 3.3.0.0 on the Combo Page there are missing some Values:
Driver Status, Last Connection to Pump, Battery, Reservoir.
With Version 3.2.* those Values were shown.

[image]

On the main page i can see the content in ml of the reservoir. That means the data is  there and i have also checked, the values are correct. So this is just a display Issue.

MilosKozak, 2025-01-01:
no logs = no help

Koeppchen, 2025-01-01:
I don't have the logs becase i have set them to Log nothing a long time ago. And also: Now i can See this Data, i don't know why. I have even tried to kill aaps to reproduce it, but directly after start the status is shown, not like in the screenshot i have added.
I don't know how to proceed, I will just close this because i can not reproduce it.

swissalpine, 2025-01-03:
It is a display problem and only affects the fields whose font color changes: In the light theme all entries are available, in the dark theme the font color is black, unless warn or urgent color are activated.
I have helped myself this way, but this is only a workaround, I have not found a real solution: I commented out all attr.defaultTextColor and the text is visible again. Here is an example (Line 116 ff.):

````
/* TODO
binding.combov2DriverState.setTextColor(
      when (connectionState) {
             ComboV2Plugin.DriverState.Error -> rh.gac(app.aaps.core.ui.R.attr.warningColor)
             ComboV2Plugin.DriverState.Suspended -> rh.gac(app.aaps.core.ui.R.attr.urgentColor)
             else -> rh.gac(app.aaps.core.ui.R.attr.defaultTextColor)
       }
)*/
```
Oder so (line 161):
```
BatteryState.FULL_BATTERY -> {
      binding.combov2Battery.text = rh.gs(R.string.combov2_battery_full_indicator)
      //binding.combov2Battery.setTextColor(rh.gac(app.aaps.core.ui.R.attr.defaultTextColor))
}
```

MilosKozak, 2025-01-04:
`app.aaps.core.ui.R.attr.defaultTextColor` is used on many places without problem

Koeppchen, 2025-01-04:
Now i do get it.
To reproduce: Set pixel phone to automatic mode for design.
Set aaps to dark background. In this case the System default Color for text is used. This is black in the case of this screenshot, because system settings are used.
@swissalpine : As a workaround i have set aaps to use the system design settings. At the moment everything works. Using the dark design is broken on Pixel with the automatic change.

swissalpine, 2025-01-04:
> `app.aaps.core.ui.R.attr.defaultTextColor` is used on many places without problem

Yes, I know, but unfortunately not in the combo tab.

> As a workaround i have set aaps to use the system design settings. At the moment everything works. Using the dark design is broken on Pixel with the automatic change.

Unfortunately, this workaround does not work for me because I use my Samsung Galaxy Flip 4 in light mode. Unfortunately, the light mode is much worse than the dark mode in AAPS (contrasts, colors), so I can't go to system settings here.
But I can personally live with my commenting out - but it will affect others who don't know how to help themselves.

MilosKozak, 2025-01-04:
hmmm it didn't work even on @jbr7rr 's initial commit https://github.com/nightscout/AndroidAPS/commit/424128d65c902d1da75a1d9ea4ae21bbf1d3885b

swissalpine, 2025-01-04:
Yes, it only changed the problem, instead of white color on white background (light mode) now black on black (dark mode).

I suspect that my workaround might be a solution if you only define the deviations urgent and warn, and otherwise leave the text color as it is.
This would only require reformulating the two when queries.
If that is a solution for you @MilosKozak, I can prepare a fix

MilosKozak, 2025-01-04:
fixed by https://github.com/nightscout/AndroidAPS/commit/69dc95cf5181942551da7b6a7524d548a1f85bc9

closed 2025-01-04 by MilosKozak