nightscout/AndroidAPS #3673: [Feature request] Upon cancelling bolus triggered by Bolus wizard, prompt to remove Carbs added by wizard for that event

GrantAdkins, 2025-01-02:
As a user, I'd like to be prompted to remove the carbs added by the bolus wizard if I cancel the bolus, so that the loop doesn't then assume COB is higher than reality, given the bolus was cancelled due to no longer eating or changing the amount being eaten.

**Example**

1. User opens Bolus wizard and enters 70g carbs for meal.
2. Bolus of 7U started.
3. User realises carb count was too high and cancels bolus after 0.05U is administered.
4. User then opens Bolus wizard again and enters the correct amount of carbs, being 50g.
5. Bolus of 5U is administered.
6. User has eaten 50g but the logged COB is 120g (70g + 50g from two events).
7. Loop administers an additional 5U over the next two hours via SMBs to handle the additional 70g that was not consumed, resulting in hypo.

gerison77, 2025-01-02:
Looks kind of the same confirmation problem of a given bolus like it is recently discussed in https://github.com/nightscout/AndroidAPS/pull/3507 . @mushroom-dev ; isn't it?

GrantAdkins, 2025-01-02:
@gerison77 it's similar, yes. The example I provided shows why some sort of prompt or warning to a user when cancelling a bolus is important. The entry of IOB and COB via a single interaction with the Bolus wizard results in the assumption that cancelling the bolus will also remove the COB entry. Of course, automatic removal of the COB entry won't apply to all scenarios, which is why I framed the user story as wanting to be prompted about what to do with the associated COB entry when the bolus is cancelled.